YosysHQ/yosys #504: can yosys infer single port block ram?

edcote, 2018-02-07:
Let's assume the following memory model generated from within [RocketChip](https://github.com/freechipsproject/rocket-chip/blob/master/scripts/vlsi_mem_gen).

I can't seem to "coerce" RocketChip to infer single-ported ram.

Is this functionality supported?

~~~~verilog
module mem_ext(
  input RW0_clk,
  input [3:0] RW0_addr, // <-- single port
  input RW0_en,
  input RW0_wmode,
  input [63:0] RW0_wdata,
  output [63:0] RW0_rdata
);

  reg reg_RW0_ren;
  reg [3:0] reg_RW0_addr;
  reg [63:0] ram [15:0];
  integer i;
  always @(posedge RW0_clk)
    reg_RW0_ren <= RW0_en && !RW0_wmode;
  always @(posedge RW0_clk)
    if (RW0_en && !RW0_wmode) reg_RW0_addr <= RW0_addr;
  always @(posedge RW0_clk)
    if (RW0_en && RW0_wmode) begin
      ram[RW0_addr][63:0] <= RW0_wdata[63:0]; // <-- write happens here
    end
  assign RW0_rdata = ram[reg_RW0_addr]; // <-- read happens here, address registered
endmodule
~~~~

Here is the simplified synthesis script:

~~~~
read_verilog -nomem2reg /mnt/raid/tmp/nade/coregen/MemCompiler/vlsi_mem_gen.v
hierarchy -check -top mem_ext
proc
memory_dff
memory_share
memory_collect
write_ilang output.ilang
~~~~

And, this is the resulting cell.  Observe how it is a dual-ported cell (i.e., separate read and write address ports).

~~~~
 cell $mem \ram
    parameter \ABITS 4
    parameter \INIT 1'x
    parameter \MEMID "\\ram"
    parameter \OFFSET 0
    parameter \RD_CLK_ENABLE 1'0
    parameter \RD_CLK_POLARITY 1'0
    parameter \RD_PORTS 1
    parameter \RD_TRANSPARENT 1'0
    parameter \SIZE 16
    parameter \WIDTH 64
    parameter \WR_CLK_ENABLE 1'0
    parameter \WR_CLK_POLARITY 1'0
    parameter \WR_PORTS 1
    connect \RD_ADDR \reg_RW0_addr
    connect \RD_CLK 1'x
    connect \RD_DATA $memrd$\ram$/mnt/raid/tmp/nade/coregen/MemCompiler/vlsi_mem_gen.v:242$104_DATA
    connect \RD_EN 1'x
    connect \WR_ADDR $memwr$\ram$/mnt/raid/tmp/nade/coregen/MemCompiler/vlsi_mem_gen.v:228$92_ADDR
    connect \WR_CLK 1'x
    connect \WR_DATA $memwr$\ram$/mnt/raid/tmp/nade/coregen/MemCompiler/vlsi_mem_gen.v:228$92_DATA
    connect \WR_EN $memwr$\ram$/mnt/raid/tmp/nade/coregen/MemCompiler/vlsi_mem_gen.v:228$92_EN
  end
~~~~

edcote, 2018-02-07:
Forgot to add `-no_rdff` to memory_dff pass.

The result is a subtle change to WR_CLK and WR_ADDR, but still dual port.

Unsupported feature?

~~~~
 cell $mem \ram
    parameter \ABITS 4
    parameter \INIT 1'x
    parameter \MEMID "\\ram"
    parameter \OFFSET 0
    parameter \RD_CLK_ENABLE 1'0
    parameter \RD_CLK_POLARITY 1'0
    parameter \RD_PORTS 1
    parameter \RD_TRANSPARENT 1'0
    parameter \SIZE 16
    parameter \WIDTH 64
    parameter \WR_CLK_ENABLE 1'1
    parameter \WR_CLK_POLARITY 1'1
    parameter \WR_PORTS 1
    connect \RD_ADDR \reg_RW0_addr
    connect \RD_CLK 1'x
    connect \RD_DATA $memrd$\ram$/mnt/raid/tmp/nade/coregen/MemCompiler/vlsi_mem_gen.v:242$104_DATA
    connect \RD_EN 1'x
    connect \WR_ADDR $procmux$113_Y
    connect \WR_CLK \RW0_clk
    connect \WR_DATA $procmux$110_Y
    connect \WR_EN $procmux$107_Y
  end
~~~~

cliffordwolf, 2018-02-08:
Support for dual port block ram in `memory_bram` is on the long-term todo list. We might assign it a higher priority soon as we are now supporting the iCE40 UltraPlus 5k in Project IceStorm (this part has ~1MBit built-in SP SRAM).

But even with this feature the `$mem` cell will still have separate read and write ports, they'd just be merged in `memory_bram`.

closed 2018-02-08 by cliffordwolf