YosysHQ/yosys #506: skip specify specparam in read_verilog

edcote, 2018-02-15:
I can't possibly be the first person who hit this?

I don't care about timing info.  Is there a switch that will force the parser to skip this section?

~~~~
Parsing Verilog input from
`[..]/RegArray16x16_m1.v' to AST representation.
ERROR: Parser error in line [..]/RegArray16x16_m1.v:235: syntax error, unexpected TOK_ID, expecting '(' or '['
~~~~

~~~~
specify
 specparam
  period_param_0 = 0.3219, // <-- HERE
  period_param_1 = 0.3505,
  period_param_2 = 0.3767,
~~~~

edcote, 2018-02-15:
Follow up: Is there a reason why the parser craps out or this has not been implemented.

I suspect the solution is in verilog_parser.y.  Never used yacc before but skipping should be trivial.  Will spend a few minutes on this and report back.

~~~~
BNF -->
specify_block ::= (From A.7.1)
specify { specify_item } endspecify
specify_item ::=
specparam_declaration
| pulsestyle_declaration
| showcancelled_declaration
| path_declaration
| system_timing_check 
~~~~

~~~~
Example -->
specify
specparam tRise_clk_q = 150, tFall_clk_q = 200;
specparam tRise_control = 40, tFall_control = 50;
endspecify
~~~~

edcote, 2018-02-16:
Well, I have a workaround.  Treat specify and endspecify as a comment.

But I am not recommending for release.

In file `verilog_lexer.l`:

~~~~
"specify" {
	register int c;
	for (;;) {
		while ((c = yyinput()) != 'e' && c != EOF);
		if (c                             == 'e') {
			if ((c = yyinput()) == 'n')
			if ((c = yyinput()) == 'd')
			if ((c = yyinput()) == 'm')
			if ((c = yyinput()) == 'o')
			if ((c = yyinput()) == 'd')
			if ((c = yyinput()) == 'u')
			if ((c = yyinput()) == 'l')
			if ((c = yyinput()) == 'e')
			return TOK_SPECIFY;
		}
		if (c == EOF)
			frontend_verilog_yyerror("endspecify keyword not found");
	}
}
~~~~

In `verilog_parser.y`:

~~~~
module_body_stmt:
	[..] | specify_decl;

specify_decl:
	TOK_SPECIFY
~~~~

TBD if you want to close this.

udif, 2018-03-05:
@edcote , please take a look at #510 . I've added support for parsing and ignoring a subset of the specify/specparam syntax, and can extend it as needed. (I wanted something out fast first)

cliffordwolf, 2018-03-27:
I've now merged the PR from @udif.

closed 2018-03-27 by cliffordwolf