YosysHQ/yosys #511: No tri-state output when assigning 1'bz

tomverbeure, 2018-03-06:
I have the following .v file:

```
module tritest(
        output my_tri_out,
        inout my_tri_inout,

        input my_in0,
        input my_in1
        );

        assign my_tri_out = 1'bz;

        assign my_tri_inout = my_in0 ? my_in1 : 1'bz;

endmodule
```

This generates the following .blif:
```
.model tritest
.inputs my_in0 my_in1
.outputs my_tri_out my_tri_inout
.names $false
.names $true
1
.names $undef
.gate $_TBUF_ A=my_in1 E=my_in0 Y=my_tri_inout
.attr src "tristate.v:12"
.names $undef my_tri_out
1 1
.end
```

When running this through arachne, I get this:
```
# arachne-pnr 0.1+284+0 (git sha1 05ea734, g++ 5.4.0-6ubuntu1~16.04.9 -O2)
.model tritest
.inputs my_in0 my_in1
.outputs my_tri_out my_tri_inout
.names $undef
.gate SB_IO PACKAGE_PIN=my_in0 LATCH_INPUT_VALUE= CLOCK_ENABLE= INPUT_CLK= OUTPUT_CLK= OUTPUT_ENABLE= D_OUT_0= D_OUT_1= D_IN_0=my_in0$2 D_IN_1=
.param PIN_TYPE 000001
.gate SB_IO PACKAGE_PIN=my_in1 LATCH_INPUT_VALUE= CLOCK_ENABLE= INPUT_CLK= OUTPUT_CLK= OUTPUT_ENABLE= D_OUT_0= D_OUT_1= D_IN_0=my_in1$2 D_IN_1=
.param PIN_TYPE 000001
.gate SB_IO PACKAGE_PIN=my_tri_inout LATCH_INPUT_VALUE= CLOCK_ENABLE= INPUT_CLK= OUTPUT_CLK= OUTPUT_ENABLE=my_in0$2 D_OUT_0=my_in1$2 D_OUT_1= D_IN_0= D_IN_1=
.param PIN_TYPE 101001
.gate SB_IO PACKAGE_PIN=my_tri_out LATCH_INPUT_VALUE= CLOCK_ENABLE= INPUT_CLK= OUTPUT_CLK= OUTPUT_ENABLE= D_OUT_0=$undef D_OUT_1= D_IN_0= D_IN_1=
.param PIN_TYPE 011001
.end
```

This issue is with the ```my_tri_out``` pin: instead of having an SB_IO with PIN_TYPE 101001 with OUTPUT_ENABLE  strapped to 0, it has an output-always-active SB_IO with PIN_TYPE 011001.

Yosys is very clear about limited support for tri-state, but the Reddit comment [here](https://www.reddit.com/r/yosys/comments/60yqrq/limited_support_for_tristate_logic/) made me think that things would be fine for a full iceStorm flow, and projects like BlackIce-II have these kind of unsupported 1'bz assignments in their examples everywhere. 

The minimal purpose of filing this issue is to have it documented so that it shows up for others when they google about it, but the better solution would be for Yosys to output something like this instead:
```
.gate $_TBUF_ A=$undef E=$false Y=my_tri_out
```

Tom

thoughtpolice, 2021-02-25:
I was looking into tristate support and stumbled across this bug; out of curiosity I figured a fix wouldn't be hard for an issue like this (which seems important), after looking around a bit, I stumbled across the long lost `insbuf` pass which almost does what we want with a small bit of fiddling:

```diff
diff --git a/passes/techmap/insbuf.cc b/passes/techmap/insbuf.cc
index a3b5b698..27fc509f 100644
--- a/passes/techmap/insbuf.cc
+++ b/passes/techmap/insbuf.cc
@@ -36,6 +36,10 @@ struct InsbufPass : public Pass {
                log("        Use the given cell type instead of $_BUF_. (Notice that the next\n");
                log("        call to \"clean\" will remove all $_BUF_ in the design.)\n");
                log("\n");
+               log("        If the celltype is $_TBUF_, <in-portname> must be A, <out-portname>\n");
+               log("        must be Y, and a disabled tri-state buffer will be inserted on all\n");
+               log("        high-impedance outputs.\n");
+               log("\n");
        }
        void execute(std::vector<std::string> args, RTLIL::Design *design) override
        {
@@ -51,6 +55,10 @@ struct InsbufPass : public Pass {
                                celltype = RTLIL::escape_id(args[++argidx]);
                                in_portname = RTLIL::escape_id(args[++argidx]);
                                out_portname = RTLIL::escape_id(args[++argidx]);
+                               if (celltype == ID($_TBUF_)) {
+                                       if (in_portname  != ID::A) log_error("<in-portname> must be 'A' for $_TBUF_ cell");
+                                       if (out_portname != ID::Y) log_error("<out-portname> must be 'Y' for $_TBUF_ cell");
+                               }
                                continue;
                        }
                        break;
@@ -70,13 +78,27 @@ struct InsbufPass : public Pass {
                                        SigBit lhs = conn.first[i];
                                        SigBit rhs = conn.second[i];

+                                       auto inserting_tbuf = celltype == ID($_TBUF_);
+                                       auto hiz_input = rhs == State::Sz;
+
                                        if (lhs.wire && !design->selected(module, lhs.wire)) {
                                                new_conn.first.append(lhs);
                                                new_conn.second.append(rhs);
                                                continue;
                                        }

+                                       if (inserting_tbuf && !hiz_input) {
+                                               // avoid cases e.g. where an input is tied
+                                               // directly to an output, which will have a buffer inserted
+                                               new_conn.first.append(lhs);
+                                               new_conn.second.append(rhs);
+                                               continue;
+                                       }
+
                                        Cell *cell = module->addCell(NEW_ID, celltype);
+                                       if (inserting_tbuf) {
+                                               cell->setPort(ID::E, State::S0);
+                                       }
                                        cell->setPort(in_portname, rhs);
                                        cell->setPort(out_portname, lhs);
                                        log("Added %s.%s: %s -> %s\n", log_id(module), log_id(cell), log_signal(rhs), log_signal(lhs));
```

This does require running `insbuf` however after your synth pass, which isn't done by default; `insbuf -buf $_TBUF_ A Y` should do the trick on your example, and the output `.blif` is what you expect: `.subckt $_TBUF_ A=$undef E=$false Y=tri_FIXME` is what I see for a hi-z output.

Of course I doubt this is the right solution that could go upstream; for example, I figured `iopadmap` might be the better place to do this, but this pass isn't used on either ecp5 or ice40 by default (my only interested targets, and only ecp5 even then), and I figured that was probably for a reason. I'm not sure what a more correct fix would look like since I'm relatively unfamiliar with the code, but I'd be willing to take a stab at it, unless this actually leads down some dark and endless pit of trouble.

widlarizer, 2026-10-13:
When you use BLIF, a format that doesn't support z-bits, and you specify a z for your output, you will be disappointed. Maybe ideally you'd get n error. I think the suggestion here is to wrap a z-bit in a Yosys-internal cell(!) as a workaround. This is a flaky convention, almost like an implicit ad-hoc extension to the BLIF format. Even the fact synth_ice40 has `_TBUF_ `survive in write_verilog output is concerning - that is not a valid ICE40 cell so it can't possibly be the result of ICE40 synthesis, and the corresponding Verilog should be emitted instead since Verilog can express that.

However I'd like to treat the issue as reported is specific to its BLIF observation. Similar unexpected behavior in other use cases should probably be under separate issues for clarity, even if they might be rooted in deeper semantic questions that would be best answered for all of RTLIL