YosysHQ/yosys #512: Assigning to a wire on the clock should be an error

ZipCPU, 2018-03-07:
The following code should generate an error:

```
module test(i_clk, o_reg);
   input wire i_clk;
   output wire o_reg;

  initial o_reg = 1'b0;
   always @(posedge i_clk)
      o_reg <= !o_reg;
endmodule
```

At issue is the fact that o_reg is a wire and not a register, therefore you should not be able to give it a value as though it were a register within an always block,

Dan

udif, 2018-03-08:
I took a look to see what's required to fix it and I've noticed that a simple fix to enforce this will break the SystemVerilog "logic" keyword which at the moment is simply aliased to "reg", and so it works as both reg/wire only because wire/reg definitions are not enforced.

~~~
module test(i_clk, o_reg, o_wire);
   input wire i_clk;
   output wire o_reg;
   output reg o_wire;

  initial o_reg = 1'b0;
   always @(posedge i_clk)
      o_reg <= !o_reg;
  assign  o_wire = !o_reg;
endmodule
~~~

I took a look at the relevant code, and it doesn't look too hard to fix. I'll try coming up with something.

udif, 2018-06-07:
PR has been updated, now passes all tests

daveshah1, 2019-05-18:
#513 is now merged, this code now produces the following warnings (that can be made into errors using `-e` if desired)
```
Warning: wire '\o_reg' is assigned in a block at wirereg.v:5.
Warning: wire '\o_reg' is assigned in a block at wirereg.v:7.
```

closed 2019-05-18 by daveshah1