YosysHQ/yosys #516: Verific Frontend Blasts Memories to DFFs

makaimann, 2018-03-23:
I've been using this simple memory as a test case: [test_mem.tar.gz](https://github.com/YosysHQ/yosys/files/1843592/test_mem.tar.gz). The memory in examples/smtbmc/demo3.v works as well but is much larger, so it is slow to run.

With the normal front-end, memories are preserved when run as follows:
```yosys -p "proc; memory -nomap; opt; prep -top test_mem -nordff; write_smt2 test_mem.smt2" -ql test_mem.log test_mem.v```

In this case, test_mem.smt2 contains Arrays.

With the Verific frontend, the memory is transformed into a bunch of DFFs. 
```yosys -p "verific -vlog2k test_mem.v; verific -import -v test_mem; proc; memory -nomap; opt; prep -top test_mem -nordff; write_smt2 verific_test_mem.smt2" -ql verific_test_mem.log```

Here, verific_test_mem.smt2 does not contain Arrays.

It appears that \mem is imported as a NetBus instead of a RamNet, which is checked for in frontends/verific.cc. Additionally, there is an instance with Type OPER_DUAL_PORT_RAM which is unhandled. I tried adding a RTLIL::Memory in that case, but got stuck trying to connect it properly.

Here's the log file for the verific case: [verific_test_mem.log](https://github.com/YosysHQ/yosys/files/1843573/verific_test_mem.log)

cliffordwolf, 2018-03-27:
I can only assume this has something to do with how verific is configured (something in VeriCompileFlags.h or maybe DBCompileFlags.h), because it works fine here with the Verific version I have. Here is the log file I get when trying to reproduce the issue: [verific_test_mem.log](https://github.com/YosysHQ/yosys/files/1852012/verific_test_mem.log) As you can see I get a memory from Verific (and an Array in my .smt2 file).

closed 2018-03-27 by cliffordwolf

makaimann, 2018-04-06:
Thanks for the quick response! That was indeed my problem. The version I'm using was configured with dualport rams instead of multiport rams as the default. This is actually configurable at runtime. I'll submit a pull request to set these runtime flags, in case other people have the same problem. As far as I know, there's no reason to use dualport over multiport rams with Yosys.