YosysHQ/yosys #517: ABC has moved.

ZipCPU, 2018-03-26:
The ABC project has [moved from bitbucket](https://bitbucket.org/alanmi/abc) to [github](https://github.com/berkeley-abc/abc).  This breaks the master yosys makefile, which references a hard URL for ABC.  Further, ABC is also no longer maintained under mercurial.  It is now under git source code control.  A quick examination suggests that only the main Makefile needs to be patched, although the patch required is a touch more intrusive than just changing URL to the new location.

cliffordwolf, 2018-03-26:
Thanks! I'll fix this right away.

mithro, 2018-03-26:
OMG! This is awesome!

cliffordwolf, 2018-03-26:
Fixed in 6f681c4.

closed 2018-03-26 by cliffordwolf

zeldin, 2018-04-30:
So maybe we can have a new release with this fix in it soon?  It's been a while since 0.7, and now it doesn't even build anymore...  :wink:

cliffordwolf, 2018-04-30:
@zeldin Uh! Yeah. That's a really good point.. I'll make a new release some time in May.