YosysHQ/yosys #524: iopadmap is confused by loops

whitequark, 2018-04-04:
Here's a design I have, which, albeit unusual, seems entirely legal to me:
```verilog
module top(
    inout  a,
    inout  b,
    input  dir,
);

    wire aout, bout;
    GP_IOBUF iobufa(
        .IO(a),
        .OUT(aout),
        .IN(bout),
        .OE(dir)
    );
    GP_IOBUF iobufb(
        .IO(b),
        .OUT(bout),
        .IN(aout),
        .OE(!dir)
    );

endmodule
```

When I run:
```
yosys top.v -p "synth_greenpak4 -part SLG46621V; show"
```
I get:
[image]
which is obviously wrong IMO.

If, instead, I alter the pipeline to only apply the iopadmap pass to `dir`, then I get:
[image]
which produces the correct post-PAR netlist.

cr1901, 2018-05-14:
I get similar results to @whitequark when trying to get the `greenpak4` backend to infer a level shifer:

Here is my input Verilog:
```
module top(
    porta0,
    portb0,
    dir,
);

(* LOC = "P2" *)
input dir;

(* LOC = "P3" *)
inout porta0;

(* LOC = "P5" *)
inout portb0;

assign porta0 = dir ? portb0 : 1'bz;
assign portb0 = ~dir ? porta0 : 1'bz;

endmodule
```

This is a minimal script which produces (what I believe to be) incorrect output. @whitequark's output can be duplicated by commenting the relevant line.

```
read_verilog level-shift/level-shift.v
hierarchy -check -top top

read_verilog -lib +/greenpak4/cells_sim.v
tribuf -logic

internal:
    # Comment the next line and run gp4-bad section to dup whitequark's output
    techmap
    clean
    show

gp4-bad:
    nlutmap -assert -luts 2,8,16,2
    iopadmap -bits -inpad GP_IBUF OUT:IN -outpad GP_OBUF IN:OUT -inoutpad GP_OBUF OUT:IN -toutpad GP_OBUFT OE:IN:OUT -tinoutpad GP_IOBUF OE:OUT:IN:IO
    techmap -map +/greenpak4/cells_map.v
    clean
    show
```

Without commenting out `techmap`, the output looks like this:
[image]

Using the following script gets me my intended design:
```
read_verilog level-shift/level-shift.v
hierarchy -check -top top

read_verilog -lib +/greenpak4/cells_sim.v
tribuf -logic

internal:
    techmap
    clean
    show

gp4:
    nlutmap -assert -luts 2,8,16,2
    select */x:dir
    iopadmap -bits -inpad GP_IBUF OUT:IN
    select */x:porta*
    iopadmap -bits -tinoutpad GP_IOBUF OE:OUT:IN:IO
    select */x:portb*
    iopadmap -bits -tinoutpad GP_IOBUF OE:OUT:IN:IO
    select -clear
    techmap -map +/greenpak4/cells_map.v
    clean
    show
```

And here is the output of `show` for the above:
[image]

cliffordwolf, 2018-05-15:
This are two different issues.

@whitequark: Fixed in  edd297f

@cr1901: Fixed in fe80b39

closed 2018-05-15 by cliffordwolf

whitequark, 2018-05-15:
Thank you! Every commit that brings Yosys to elimination of the "tristate support is experimental" warning warms my heart.