YosysHQ/yosys #525: yosys-smtbmc: Error when using "--stbv" option

blazra, 2018-04-04:
I am using symbiyosys and when I add --stbv option to the smtbmc engine it crashes. Without the option it seems to work fine. (I am not sure what the option does, I was just trying it out, but this doesn't seem to be correct behavior)
With yices it says:
>File "/usr/local/bin/yosys-smtbmc", line 392, in <module>
smt.write(line)
File "/usr/local/bin/../share/yosys/python3/smtio.py", line 409, in write
self.p_write(stmt + "\n", True)
File "/usr/local/bin/../share/yosys/python3/smtio.py", line 293, in p_write
if flush: self.p.stdin.flush()
BrokenPipeError: [Errno 32] Broken pipe

and with z3:
>Solver Error: (error "line 564 column 64: invalid declaration, named expression already defined with this name testbench_h uut")

Complete outputs:
With yices:

> SBY 15:21:14 [trigger_rules] Removing direcory 'trigger_rules'.
SBY 15:21:14 [trigger_rules] Copy 'trigger_rules.vhd' to 'trigger_rules/src/trigger_rules.vhd'.
SBY 15:21:14 [trigger_rules] Copy 'formal_tb.sv' to 'trigger_rules/src/formal_tb.sv'.
SBY 15:21:14 [trigger_rules] engine_0: smtbmc --stbv yices
SBY 15:21:14 [trigger_rules] base: starting process "cd trigger_rules/src; yosys -ql ../model/design.log ../model/design.ys"
SBY 15:21:14 [trigger_rules] base: -- (c) Copyright 1999 - 2018 Verific Design Automation Inc. All rights reserved
SBY 15:21:14 [trigger_rules] base: -- Compilation time was Thu Mar 29 21:26:25 2018
SBY 15:21:14 [trigger_rules] base: -- This program will expire on Sat Jul  7 21:26:25 2018
SBY 15:21:24 [trigger_rules] base: finished (returncode=0)
SBY 15:21:24 [trigger_rules] smt2_stbv: starting process "cd trigger_rules/model; yosys -ql design_smt2_stbv.log design_smt2_stbv.ys"
SBY 15:21:24 [trigger_rules] smt2_stbv: -- (c) Copyright 1999 - 2018 Verific Design Automation Inc. All rights reserved
SBY 15:21:24 [trigger_rules] smt2_stbv: -- Compilation time was Thu Mar 29 21:26:25 2018
SBY 15:21:24 [trigger_rules] smt2_stbv: -- This program will expire on Sat Jul  7 21:26:25 2018
SBY 15:21:24 [trigger_rules] smt2_stbv: finished (returncode=0)
SBY 15:21:24 [trigger_rules] engine_0.basecase: starting process "cd trigger_rules; yosys-smtbmc -s yices --presat --unroll --noprogress -t 30 --append 0 --dump-vcd engine_0/trace.vcd --dump-vlogtb engine_0/trace_tb.v --dump-smtc engine_0/trace.smtc model/design_smt2_stbv.smt2"
SBY 15:21:24 [trigger_rules] engine_0.induction: starting process "cd trigger_rules; yosys-smtbmc -s yices --presat --unroll -i --noprogress -t 30 --append 0 --dump-vcd engine_0/trace_induct.vcd --dump-vlogtb engine_0/trace_induct_tb.v --dump-smtc engine_0/trace_induct.smtc model/design_smt2_stbv.smt2"
SBY 15:21:24 [trigger_rules] engine_0.basecase: ##   0:00:00  Solver: yices
SBY 15:21:24 [trigger_rules] engine_0.induction: ##   0:00:00  Solver: yices
SBY 15:21:25 [trigger_rules] engine_0.induction: Traceback (most recent call last):
SBY 15:21:25 [trigger_rules] engine_0.induction: File "/usr/local/bin/yosys-smtbmc", line 392, in <module>
SBY 15:21:25 [trigger_rules] engine_0.induction: smt.write(line)
SBY 15:21:25 [trigger_rules] engine_0.induction: File "/usr/local/bin/../share/yosys/python3/smtio.py", line 409, in write
SBY 15:21:25 [trigger_rules] engine_0.induction: self.p_write(stmt + "\n", True)
SBY 15:21:25 [trigger_rules] engine_0.induction: File "/usr/local/bin/../share/yosys/python3/smtio.py", line 293, in p_write
SBY 15:21:25 [trigger_rules] engine_0.induction: if flush: self.p.stdin.flush()
SBY 15:21:25 [trigger_rules] engine_0.induction: BrokenPipeError: [Errno 32] Broken pipe
SBY 15:21:25 [trigger_rules] engine_0.basecase: Traceback (most recent call last):
SBY 15:21:25 [trigger_rules] engine_0.basecase: File "/usr/local/bin/yosys-smtbmc", line 392, in <module>
SBY 15:21:25 [trigger_rules] engine_0.basecase: smt.write(line)
SBY 15:21:25 [trigger_rules] engine_0.basecase: File "/usr/local/bin/../share/yosys/python3/smtio.py", line 409, in write
SBY 15:21:25 [trigger_rules] engine_0.basecase: self.p_write(stmt + "\n", True)
SBY 15:21:25 [trigger_rules] engine_0.basecase: File "/usr/local/bin/../share/yosys/python3/smtio.py", line 293, in p_write
SBY 15:21:25 [trigger_rules] engine_0.basecase: if flush: self.p.stdin.flush()
SBY 15:21:25 [trigger_rules] engine_0.basecase: BrokenPipeError: [Errno 32] Broken pipe
SBY 15:21:25 [trigger_rules] engine_0.basecase: finished (returncode=1)
Traceback (most recent call last):
  File "/usr/local/bin/sby", line 262, in <module>
    retcode |= run_job(t)
  File "/usr/local/bin/sby", line 248, in run_job
    job.run()
  File "/usr/local/bin/../share/yosys/python3/sby_core.py", line 524, in run
    self.taskloop()
  File "/usr/local/bin/../share/yosys/python3/sby_core.py", line 192, in taskloop
    task.poll()
  File "/usr/local/bin/../share/yosys/python3/sby_core.py", line 115, in poll
    self.handle_exit(self.p.returncode)
  File "/usr/local/bin/../share/yosys/python3/sby_core.py", line 71, in handle_exit
    self.exit_callback(retcode)
  File "/usr/local/bin/../share/yosys/python3/sby_engine_smtbmc.py", line 149, in exit_callback
    assert task_status is not None
AssertionError

With z3:
>SBY 15:28:32 [trigger_rules] Removing direcory 'trigger_rules'.
SBY 15:28:32 [trigger_rules] Copy 'trigger_rules.vhd' to 'trigger_rules/src/trigger_rules.vhd'.
SBY 15:28:32 [trigger_rules] Copy 'formal_tb.sv' to 'trigger_rules/src/formal_tb.sv'.
SBY 15:28:32 [trigger_rules] engine_0: smtbmc --stbv z3
SBY 15:28:32 [trigger_rules] base: starting process "cd trigger_rules/src; yosys -ql ../model/design.log ../model/design.ys"
SBY 15:28:32 [trigger_rules] base: -- (c) Copyright 1999 - 2018 Verific Design Automation Inc. All rights reserved
SBY 15:28:32 [trigger_rules] base: -- Compilation time was Thu Mar 29 21:26:25 2018
SBY 15:28:32 [trigger_rules] base: -- This program will expire on Sat Jul  7 21:26:25 2018
SBY 15:28:42 [trigger_rules] base: finished (returncode=0)
SBY 15:28:42 [trigger_rules] smt2_stbv: starting process "cd trigger_rules/model; yosys -ql design_smt2_stbv.log design_smt2_stbv.ys"
SBY 15:28:42 [trigger_rules] smt2_stbv: -- (c) Copyright 1999 - 2018 Verific Design Automation Inc. All rights reserved
SBY 15:28:42 [trigger_rules] smt2_stbv: -- Compilation time was Thu Mar 29 21:26:25 2018
SBY 15:28:42 [trigger_rules] smt2_stbv: -- This program will expire on Sat Jul  7 21:26:25 2018
SBY 15:28:43 [trigger_rules] smt2_stbv: finished (returncode=0)
SBY 15:28:43 [trigger_rules] engine_0: starting process "cd trigger_rules; yosys-smtbmc -s z3 --presat --noprogress -t 30 --append 0 --dump-vcd engine_0/trace.vcd --dump-vlogtb engine_0/trace_tb.v --dump-smtc engine_0/trace.smtc model/design_smt2_stbv.smt2"
SBY 15:28:43 [trigger_rules] engine_0: ##   0:00:00  Solver: z3
SBY 15:28:43 [trigger_rules] engine_0: ##   0:00:00  Checking assumptions in step 0..
SBY 15:28:43 [trigger_rules] engine_0: ##   0:00:00  Solver Error: (error "line 564 column 64: invalid declaration, named expression already defined with this name testbench_h uut")
SBY 15:28:44 [trigger_rules] engine_0: finished (returncode=1)
Traceback (most recent call last):
  File "/usr/local/bin/sby", line 262, in <module>
    retcode |= run_job(t)
  File "/usr/local/bin/sby", line 248, in run_job
    job.run()
  File "/usr/local/bin/../share/yosys/python3/sby_core.py", line 524, in run
    self.taskloop()
  File "/usr/local/bin/../share/yosys/python3/sby_core.py", line 192, in taskloop
    task.poll()
  File "/usr/local/bin/../share/yosys/python3/sby_core.py", line 115, in poll
    self.handle_exit(self.p.returncode)
  File "/usr/local/bin/../share/yosys/python3/sby_core.py", line 71, in handle_exit
    self.exit_callback(retcode)
  File "/usr/local/bin/../share/yosys/python3/sby_engine_smtbmc.py", line 149, in exit_callback
    assert task_status is not None
AssertionError

cliffordwolf, 2018-04-04:
You do not provide a test case that I could use to reproduce the problem. So there is really nothing I can do. Therefore I'm closing the issue. I'll reopen the issue if and when a test case is provided. (See also Q1 in the [FAQ](http://www.clifford.at/yosys/faq.html).)

cliffordwolf, 2018-04-04:
Thanks for the test case. Fixed in 25a864f.

closed 2018-04-04 by cliffordwolf