YosysHQ/yosys #532: Yosys doesn't fail on invalid argument given to abc command

mithro, 2018-04-15:
## Steps to reproduce the issue

I was trying to run the following command;
```
yosys -p "synth; abc -lut 4; opt_clean; write_blif -attr -cname -conn -impltf -param build/testarch/2x4/vlut.eblif" vlut.v
```

But accidentally screwed up and had the following command line;
```
yosys -p "synth; abc -lut 4 opt_clean; write_blif -attr -cname -conn -impltf -param build/testarch/2x4/vlut.eblif" vlut.v
```

## Expected behavior

Yosys fails with an error.

## Actual behavior

Yosys runs and doesn't seem to complain in any way? 

As far as I can tell, `opt_clean` isn't a valid argument to the `abc` command. There isn't anything named `opt_clean` inside my verilog to be selected either?

The log output looks like;
```
 /----------------------------------------------------------------------------\
 |                                                                            |
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |                                                                            |
 |  Copyright (C) 2012 - 2018  Clifford Wolf <clifford@clifford.at>           |
 |                                                                            |
 |  Permission to use, copy, modify, and/or distribute this software for any  |
 |  purpose with or without fee is hereby granted, provided that the above    |
 |  copyright notice and this permission notice appear in all copies.         |
 |                                                                            |
 |  THE SOFTWARE IS PROVIDED "AS IS" AND THE AUTHOR DISCLAIMS ALL WARRANTIES  |
 |  WITH REGARD TO THIS SOFTWARE INCLUDING ALL IMPLIED WARRANTIES OF          |
 |  MERCHANTABILITY AND FITNESS. IN NO EVENT SHALL THE AUTHOR BE LIABLE FOR   |
 |  ANY SPECIAL, DIRECT, INDIRECT, OR CONSEQUENTIAL DAMAGES OR ANY DAMAGES    |
 |  WHATSOEVER RESULTING FROM LOSS OF USE, DATA OR PROFITS, WHETHER IN AN     |
 |  ACTION OF CONTRACT, NEGLIGENCE OR OTHER TORTIOUS ACTION, ARISING OUT OF   |
 |  OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.            |
 |                                                                            |
 \----------------------------------------------------------------------------/

 Yosys 0.7+501 (git sha1 588ce0e3, x86_64-conda_cos6-linux-gnu-gcc 8a21ab48 -fPIC -Os)

-- Parsing `vlut.v' using frontend `verilog' --

1. Executing Verilog-2005 frontend.
Parsing Verilog input from `vlut.v' to AST representation.
Generating RTLIL representation for module `\top'.
Note: Assuming pure combinatorial block at vlut.v:3 in
compliance with IEC 62142(E):2005 / IEEE Std. 1364.1(E):2002. Recommending
use of @* instead of @(...) for better match of synthesis and simulation.
Successfully finished Verilog frontend.

-- Running command `synth; abc -lut 4 opt_clean; write_blif -attr -cname -conn -impltf -param build/testarch/2x4/vlut.eblif' --

2. Executing SYNTH pass.

2.1. Executing HIERARCHY pass (managing design hierarchy).

2.2. Executing PROC pass (convert processes to netlists).

2.2.1. Executing PROC_CLEAN pass (remove empty switches from decision trees).
Cleaned up 0 empty switches.

2.2.2. Executing PROC_RMDEAD pass (remove dead branches from decision trees).
Removed a total of 0 dead cases.

[... 421 lines cut ...]
3. Executing ABC pass (technology mapping using ABC).

4. Executing BLIF backend.

End of script. Logfile hash: 8d4e70a2d7
CPU: user 0.03s system 0.00s, MEM: 37.84 MB total, 9.77 MB resident
Yosys 0.7+501 (git sha1 588ce0e3, x86_64-conda_cos6-linux-gnu-gcc 8a21ab48 -fPIC -Os)
Time spent: 37% 12x opt_expr (0 sec), 24% 7x opt (0 sec), ...
```

```verilog
// 4-input LUT test.
module top(input [3:0] I, output O);
    always @(I)
    case(I)
        4'b1000 : O = 1;
        4'b1001 : O = 1;
        4'b1010 : O = 1;
        4'b1100 : O = 1;
        4'b1110 : O = 1;
        4'b1111 : O = 1;
        default : O = 0;
    endcase
endmodule // top
```

mithro, 2018-04-15:
Is selecting something which doesn't exist a normal part of standard yosys flows? I'm guessing you're using it rather than "if x exists do y"?

cliffordwolf, 2018-04-16:
See slides 70/150 - 79/150:
http://www.clifford.at/yosys/files/yosys_presentation.pdf

Or pages 3ff in this document:
http://www.clifford.at/yosys/files/yosys_appnote_011_design_investigation.pdf

cliffordwolf, 2018-04-16:
Or just `help select` of course:   :)
http://www.clifford.at/yosys/cmd_select.html

The second paragraph of `help select` mentions `[selection]` that many (most) commands support. For example `abc`:
http://www.clifford.at/yosys/cmd_abc.html

mithro, 2018-04-17:
@cliffordwolf - After reading that it is still unclear to me if `select <name of thing which doesn't exist>` is considered a valid thing to be doing and if it should be a regular occurrence during normal Yosys operation?

IE Is it reasonable to expect yosys to warn/error about selection of things that don't exist or is selection of things that don't exist a normal behaviour?