YosysHQ/yosys #533: When should yosys be generating extended eblif attributes?

mithro, 2018-04-15:
## Steps to reproduce the issue

Running the following verilog file through Yosys;
```verilog
// 4-input LUT test.
module top(input [3:0] I, output O);
    always @(I)
    case(I)
        4'b1000 : O = 1;
        4'b1001 : O = 1;
        4'b1010 : O = 1;
        4'b1100 : O = 1;
        4'b1110 : O = 1;
        4'b1111 : O = 1;
        default : O = 0;
    endcase
endmodule // top
```

With the following command `synth_ice40; ice40_opt -unlut; abc -lut 4; opt_clean; write_blif -attr -cname -conn -impltf -param vlut.eblif` you get;
```blif
# Generated by Yosys 0.7+501 (git sha1 588ce0e3, x86_64-conda_cos6-linux-gnu-gcc 8a21ab48 -fPIC -Os)

.model top
.inputs I[0] I[1] I[2] I[3]
.outputs O
.names I[3] I[2] I[1] I[0] O
1000 1
1001 1
1010 1
1100 1
1110 1
1111 1
.end
```

If you instead use the following command `synth_ice40; write_blif -attr -cname -conn -impltf -param vlut.eblif` you get;
```blif
# Generated by Yosys 0.7+501 (git sha1 588ce0e3, x86_64-conda_cos6-linux-gnu-gcc 8a21ab48 -fPIC -Os)

.model top
.inputs I[0] I[1] I[2] I[3]
.outputs O
.subckt SB_LUT4 I0=I[0] I1=I[1] I2=I[2] I3=I[3] O=O
.cname $abc$173$auto$blifparse.cc:465:parse_blif$174
.param LUT_INIT 1101011100000000
.end
```

## Expected behavior

I would have expected a `.cname` for the `.names` in the first example?

I would have also expected in both cases something which documents where in the Verilog file this lut comes from?

cliffordwolf, 2018-04-15:
> I would have expected a .cname for the .names in the first example?

Up until now `.cname`, `.attr`, and `.param` are only generated for cells, not formulas. (You can write the BLIF file with `-icells` if you wanted cells in the output for built-in cell types instead of formulas.)

In 5ca91ca I've now added the options `-iname` and `-iattr` that can be used to enable `.cname` and `.attr` on `.names` blocks. (Enabling `.param` does not make any sense for `.names`.)

> I would have also expected in both cases something which documents where in the Verilog file this lut comes from?

Then you must not use logic optimization. We can not preserve any information on inner points of a logic function. (How would that even work? There is no 1:1 relation between cells before and after logic optimization. Everything is translated into a huge boolean formula and then a new circuit is generated from that formula.)

closed 2018-04-15 by cliffordwolf

mithro, 2018-04-15:
> Up until now .cname, .attr, and .param are only generated for cells, not formulas. (You can write the BLIF file with -icells if you wanted cells in the output for built-in cell types instead of formulas.)

By "formulas" you mean `.names`? Does that include `.latches` too?

I guess I didn't really think there was a difference between any of the "things" which start with a `.`. 

> In 5ca91ca I've now added the options -iname and -iattr that can be used to enable .cname and .attr on .names blocks. (Enabling .param does not make any sense for .names.)

Does it make sense to to have `.attr` and `.param` or similar things on `.input` and `.outputs` or other similar nets?

> Then you must not use logic optimization. We can not preserve any information on inner points of a logic function. (How would that even work? There is no 1:1 relation between cells before and after [...]

Should we move this part to a separate meta bug for further discussion?

I haven't thought about this really hard, but what I would be awesome to see as the final result is from input like this;
```verilog
// 4-input LUT test.                    // Line 1
module top(input [3:0] I, output O);    // Line 2
always @(I)				// Line 3
    case(I)                             // Line 4
        4'b1000 : O = 1;                // Line 5
        4'b1001 : O = 1;                // Line 6
        4'b1010 : O = 1;                // Line 7
        4'b1100 : O = 1;                // Line 8
        4'b1110 : O = 1;                // Line 9
        4'b1111 : O = 1;                // Line 10
        default : O = 0;                // Line 11
    endcase                             // Line 12
endmodule // top                        // Line 13
```

You would somehow get an annotation kind of like this;
```blif
.model top                    # file:lut.v@line:2@col:8
.inputs I[0] I[1] I[2] I[3]   # file:lut.v@line:2@col:24
.outputs O                    # file:lut.v@line:2@col:34
.names I[3] I[2] I[1] I[0] O  # file:lut.v@line:3@col:0
1000 1                        # file:lut.v@line:5@col:9
1001 1                        # file:lut.v@line:6@col:9
1010 1                        # file:lut.v@line:7@col:9
1100 1                        # file:lut.v@line:8@col:9
1110 1                        # file:lut.v@line:9@col:9
1111 1                        # file:lut.v@line:10@col:9
.end
```

The way I *think* you could make this work is to tag every value and operation with the attributes. When you expand something, you copy these attributes to the outputs. When you reduce something, you merge the attributes values.

I'm guessing that some values will potentially end up being tagged with like 400 attribute lines...

I'm guessing it would probably be very hard to preserve this information through an ABC pass without modifications to that?

cliffordwolf, 2018-04-16:
> Should we move this part to a separate meta bug for further discussion?

I don't think there is a bug here, meta or otherwise.

Yosys preserved all relevant information up until the point where you instructed it to rip out all logic and replace it with entirely new logic by calling ABC. That's what it did. Don't want that? Then don't call ABC.

If you want to preserve a *wire* (not cell) then simply set the `keep` attribute on it. Then the `abc` pass will treat that wire like an output and preserve it, at the cost of a potentially less optimal circuit. But the whole concept of "preserving information attached to cells" when processing a logic formula with ABC is simply nonsensical.

> I'm guessing it would probably be very hard to preserve this information through an ABC pass without modifications to that?

Apparently I'm not expressing that clearly enough: There is inherently NO WAY to "pass information attached to cells through ABC". For the purpose of this discussion calling ABC is equivalent to generating an enormous monolithic thruth table for all your logic and then synthesizing an entirely new circuitry from that. There is no clear relation between cells in the ABC input netlist and cells in the ABC output netlist. This is not because someone forgot to add a feature but because of the kind of algorithms that ABC uses.

You can of course write a different logic optimization tool, using entirely different algorithms and data structures, that behaves exactly the way you want. But that's independent of Yosys and the absence of such a tool does not in any way constitute a bug!

I'd be happy to help integrating such a program with Yosys once you have written it.