YosysHQ/yosys #536: Compile yosys fail .....

red0bear, 2018-04-17:
i tried to compile but it give me this .... i wish know what is happen ....

[ 11%] Building frontends/ilang/ilang_frontend.o
[ 12%] Building frontends/json/jsonparse.o
[ 12%] Building frontends/verilog/verilog_parser.tab.cc
frontends/verilog/verilog_parser.y:142.9-19: error: %define variable 'parse.error' is not used
frontends/verilog/Makefile.inc:8: recipe for target 'frontends/verilog/verilog_parser.tab.cc' failed

daveshah1, 2018-04-22:
This issue came up previously: https://www.reddit.com/r/yosys/comments/6j163z/parseerror_error_in_yosys_installation/

Try and update Bison and see if that helps.

red0bear, 2018-04-22:
i see this after i make update to the last Bison. You can close this if you wish.

d235j, 2018-05-04:
@red0bear what version of Bison are you using?

red0bear, 2018-05-04:
@d235j under last opensuse leap Bison official version is 2.7.11. I just updated to version 3.0.4 and it worked. I wish bison constructor could at least have support to old version. But now its ok. I would like people help me out with another question i had made here last week if possible. I Have a project and need work on it using this tool.

closed 2019-05-18 by daveshah1