YosysHQ/yosys #539: Calculate a min typ max using .lib

red0bear, 2018-04-23:
I was studying how calculate these values to give a approach about gate level simulation using yosys to generate a verilog under cell libs on vsclib013.lib . But you need STA time calculation Path to do that. So im here thinking in how could do a  simple sample to help me at least to estimate this. 

here a example to after generate the cell libs in a verilog file.  

module an2v0x05(a, b, z);

  input a;
  input b;
  output z;
  assign #(min:typ:max;min:typ:max;min:typ:max)z =  a & b;

endmodule

byzantic, 2018-06-04:
It's unclear what you actually want to do..

I _think_ that you want to be able to to a full timing simulation using a netlist that has been generated by synthesising a design using Yosys. That will generate a netlist, but there is no timing information.

I think you are saying that the cell library (probably vsclib013.v?) does not contain any timing information? That's because, although at one time, timing _was_ included in verilog cell libraries, it is insufficient to perform accurate timing simulations. A large part of the timing delays is now due to the load dependent delays, and this _cannot_ be modelled using just a cell library; each instance of a cell will have different delays depending on how many cells and wiring it is connected to.

To perform timing simulation, you need the layout tool to generate a timing annotation file for the simulator. This is known as an SDF ( .sdf) file, and it contains timing for each gate instance.

The open source simulators GPL cver and cvc can use .sdf files.

.. But you probably don't have access to a layout tool, and no way to generate an SDF file!

However, most design teams only use timing simulation as a final check that the design is OK. Design work is usually carried out using STA.

.. and there is an open source STA tool available called OpenTimer (see https://web.engr.illinois.edu/~thuang19/software/timer/OpenTimer.html )

The problem here is that OpenTimer needs some more data. It needs the cell library files (.lib),  timing definitions (which you will have to write), and layout capacitances, resistance, etc. in a .spef file

.spef files are generated from a layout tool, so you won't have access to them. However, you can write an empty file that just includes the header and has no information about circuit nodes. OpenTimer will accept that.

Annoyingly, it also wants two .lib files, an early and a late version. However, most .lib files (such as vsclib013.lib contain one set of data (BUT they also contain the rules about how to derate for temperature, wire loads etc ..)

I haven't tried yet, but I think it should be possible to just give the same .lib file for both early and late, and that should be OK. Have a play with OpenTimer!

I suggest you close this issue, because it isn't really an issue.

What _would_ be really great is to find some way of doing the following:

1. generate an SDF file for a synthesised netlist, using early/typ/late timing AND a wire load model 
    all the information is available in the .lib files already, and Yosys already does the calculations for the 
    -D option

    this would allow verilog timing simulations, using estimated wire loads.

2. generate an (estimated) SPEF file 
    I _think_ the .lib file contains enough information to generate this; effectively it's the first part of (1)

   This would allow OpenTimer to run on the netlist generated by Yosys.

  This could also be a stand-alone program. Anyone know of anything suitable?

3. Generate specialised .lib files from a generic .lib file (i.e. generate early/late .lib files for OpenTimer)

    This isnt really a feature for Yosys, since Yosys handles .lib files OK.

red0bear, 2018-06-04:
Give me a time to analyze your response. Anyway thank you.

red0bear, 2018-07-05:
hello man .... after have think about you write seems we have good options but for what i want. 
Yosys for me is like RTLCompiler ... but with a few options. OpenTimer seems good but with no enough info to work with icarus verilog like yosys. Was very hard make SystemC works with Icarus verilog with no docs ..... but i there is a way ........ Well i consider this topic like a missing feature.  

I'm using yosys on this project https://github.com/GLADICOS/SPACEWIRESYSTEMC
i wish to test yosys Altera build or another one output on icarus verilog with gate level to determine problem where i could not see simulating in zero delay.  no need be sdf file .... but XML or a JSON file compatible to load in icarus .