YosysHQ/yosys #541: running external abc in yosys

pasandi20, 2018-05-02:
## Steps to reproduce the issue

*Provide instructions for reproducing the issue. Make sure to include
all neccessary source files. (You can simply drag&drop a .zip file into
the issue editor.)*

## Expected behavior

*Please describe the behavior you would have expected from the tool.*

## Actual behavior

*Please describe how the behavior you see differs from the expected behavior.*

**Important Note:** Nobody will be able to help you and/or fix the issue if you
do not provide sufficient information for reproducing the problem.

pasandi20, 2018-05-02:
In the MakeFile of Yosys, there is an option for selecting an external ABC. It is described as follows:

# set ABCEXTERNAL = <abc-command> to use an external ABC instance
# Note: The in-tree ABC (yosys-abc) will not be installed when ABCEXTERNAL is set.
ABCEXTERNAL ?=

My understanding is that I should edit the last line and give a path to the external exe of ABC, something like this:

ABCEXTERNAL = ~abc-master/abc.exe

However, when I compile the code and then run it, it seems like the tool is ignoring the external exe of abc. It is just looking for abc-default.exe or yosys-abc.exe in the directory of yosys and if it cannot find them it returns a corresponding error (that there is no abc!)

cliffordwolf, 2018-05-02:
What does it say in the console output? (Under the "Executing ABC." header.)

Did you run a clean build after editing the Makefile?

pasandi20, 2018-05-02:
6.1.1. Executing ABC.
Running ABC command: ~abc-master/abc.exe -s -f <abc-temp-dir>/abc.script 2>&1
ABC: sh: 1: ~abc-master/abc.exe: not found
ERROR: ABC: execution of command "~abc-master/abc.exe -s -f /tmp/yosys-abc-kVFTNB/abc.script 2>&1" failed: return code 127.

cliffordwolf, 2018-05-02:
See! It worked. All you have to do is provide a valid path name to your ABC binary.

closed 2018-05-02 by cliffordwolf

pasandi20, 2018-05-02:
Something that I'v see which looks like a bug is that even if I change the path in front of ABCEXTERNAL above, and re-make the yosys code, it will ignore the changes, unless for example I change also something in the source code of yosys (such as passes/techmap/abc.cc)

cliffordwolf, 2018-05-02:
That's not a bug. You have changed the Makefile. There is no make dependency for that. (The only meaningful would be to rebuild everything anyways.) So you have to run a clean build. Same as with pretty much every other Makefile based project.

pasandi20, 2018-05-02:
I got your point but cleaning and re-building is not very convenient because we have to make it again and it is time consuming. Thanks