YosysHQ/yosys #544: Using  optimization scripts of ABC such as "resyn" in Yosys

pasandi20, 2018-05-02:
## Steps to reproduce the issue

*Provide instructions for reproducing the issue. Make sure to include
all neccessary source files. (You can simply drag&drop a .zip file into
the issue editor.)*

## Expected behavior

*Please describe the behavior you would have expected from the tool.*

## Actual behavior

*Please describe how the behavior you see differs from the expected behavior.*

**Important Note:** Nobody will be able to help you and/or fix the issue if you
do not provide sufficient information for reproducing the problem.

pasandi20, 2018-05-02:
I tried to add those scripts on the set of commands being passed to ABC (which are determined in passes/techmap/abc.cc), but it was not successful, I received the following error on console:

ABC: + resyn 
ABC: ** cmd error: unknown command 'resyn'
ABC: (this is likely caused by using an alias defined in "abc.rc"
ABC: without having this file in the current or parent directory)
ABC: ** cmd error: aborting 'source <abc-temp-dir>/abc.script'
ERROR: Can't open ABC output file `/tmp/yosys-abc-HTYpnh/output.blif'.

I copied the abc.rc on both the directory of yosys and the abc which I'm using, but the above problem is still there.

cliffordwolf, 2018-05-04:
PS: Unfortunately you have not posted the part of the log file where you can see the ABC command line. From your previous issue:

> 6.1.1. Executing ABC.
> Running ABC command: ~abc-master/abc.exe -s -f /abc.script 2>&1

As you can see `abc` is being called with option `-s`. From the `abc -h` help message:

>    -s		do not read any initialization file

So no `abc.rc` is being read. This is intentional.