YosysHQ/yosys #545: `check` does not appear to detect many cases of multiple drivers

whitequark, 2018-05-04:
## Steps to reproduce the issue

Verilog code:
```verilog
module top(inout a, input b);
  always @(*) a <= 1'b0;
  always @(*) a <= b ? 1'b0 : 1'b1;
  // or:
  // always @(*) a <= b;
endmodule
```
Yosys script:
```
proc
check -assert
```

## Expected behavior

`check` would log an error.

## Actual behavior

```
3. Executing CHECK pass (checking for obvious problems).
checking module top..
found and reported 0 problems.
```

## Notes

`opt_clean` prints a warning for the uncommmented case but not the commented one.

Based on my understanding of IEEE 1364.1-2002, both of the cases above are not legal synthesizable Verilog:
[image]

azonenberg, 2023-06-23:
See #3817

jix, 2023-06-26:
Now with #3817 the conflict in the example is detected. I'm not aware of any other cases where we miss multiple drivers.

closed 2023-06-26 by jix