YosysHQ/yosys #547: `stat` does not appear to output anything when -v or -q is supplied

thoughtpolice, 2018-05-11:
## Steps to reproduce the issue

With any design at all, write a `.ys` or `.tcl` script that simply synthesizes (or really does anything) and runs `stat` at the end, to output cell information about the selected hierarchy. Next, invoke `yosys` with an argument like `-v2`, and watch how `stat` does nothing. Here is a sample script:

```
read_verilog *.v

hierarchy; proc; fsm; opt; memory; opt
techmap; opt;

stat
```

At the end during the `STAT` pass, there will be no information output at all. This is because `-v` implies `-q` at all levels, which only results in warnings or errors being output.

## Expected behavior

I would expect `stat` to always output information about the design, **regardless of if I am executing quietly or at a particular verbosity level**.

This is simply because *I literally asked* for the output, by inserting the command explicitly! If there is some implicit usage of `stat`, I would like these to respect the verbosity/quiet settings, but otherwise.

*Please describe the behavior you would have expected from the tool.*

## Actual behavior

`stat` returns no output in quiet mode.

## Motivation

I have a `.tcl` script which drives `yosys` on a large design in order to report cell count of the final synthesis results -- but I don't care about most of the passes or their output (especially the nested logs). I only care about the stats at the end, more or less. Synthesis takes a few minutes -- thus, I would simply like to run `yosys` at `-v1` to give a "high level overview" of where the compilation is at a glance while waiting, and still get the stats at the end. (As a happy coincidence, running at `-v1` produces relatively little output to where I can also trivially see any warnings very easily; I could use `-w` for this, but it's just a nice side bonus of the output being way shorter!)

thoughtpolice, 2018-05-11:
I am also willing to write a patch if my motivation/stance is convincing. :)

cliffordwolf, 2018-05-12:
This is the intended behavior. In `-v2` mode only headers of the first two "indentation levels" and warnings and errors are printed. All log messages are hidden in that mode.

Use something like `tee -o /dev/stdout stat` if you want to make the stat output appear on stdout.

closed 2018-05-12 by cliffordwolf

thoughtpolice, 2018-05-13:
Ah, thanks! I completely missed `tee` in the listing, but that's a good trick!