YosysHQ/yosys #548: make test fails

moohtuh, 2018-05-13:
Hello! I get this error: 

make[1]: Entering directory '/home/muhtadic16/Downloads/yosys-yosys-0.7/tests/simple'
Test: aes_kexp128 -> ERROR!
Note: Make sure that 'iverilog' is an up-to-date git checkout of Icarus Verilog.
../tools/autotest.mk:10: recipe for target 'aes_kexp128.v' failed
make[1]: *** [aes_kexp128.v] Error 1
make[1]: Leaving directory '/home/muhtadic16/Downloads/yosys-yosys-0.7/tests/simple'
Makefile:416: recipe for target 'test' failed
make: *** [test] Error 2

My Iverilog is updated already (but still keep getting the error) and I bypassed installing ABC while doing the "make" (may be that is a reason for the error?).  

Also, it shows "build successful" upon giving the 'make' command.

Please help.

moohtuh, 2018-05-14:
Anybody help please?

cliffordwolf, 2018-05-14:
The failing test must have produced an error log file. What does it say?

moohtuh, 2018-05-14:
It says :       

6.2. Extracting gate netlist of module `\aes_key_expand_128' to `<abc-temp-dir>/input.blif'..
No matching clock domain found. Not extracting any FF cells.
Extracted 28 gates and 61 wires to a netlist network with 33 inputs and 16 outputs.

6.2.1. Executing ABC.
ERROR: ABC: execution of command "/home/muhtadic16/Downloads/yosys-yosys-0.7/yosys-abc -s -f /tmp/yosys-abc-GMlkMP/abc.script 2>&1" failed: return code 127.

cliffordwolf, 2018-05-14:
Return code 127 is the exec() exit code for command not found.

> I bypassed installing ABC while doing the "make" (may be that is a reason for the error?).

Don't do that then.

closed 2018-05-14 by cliffordwolf

moohtuh, 2018-05-14:
Also when I keep the makefile unedited this happens :  

muhtadic16@ubuntu:~/Downloads/yosys-yosys-0.7$ make
[  0%] Building abc/abc-eb6eca6807cc
Pulling ABC from https://bitbucket.org/alanmi/abc:
+ test -d abc
+ hg clone https://bitbucket.org/alanmi/abc abc
/bin/sh: 4: hg: not found
+ cd abc
/bin/sh: 5: cd: can't cd to abc
Makefile:387: recipe for target 'abc/abc-eb6eca6807cc' failed
make: *** [abc/abc-eb6eca6807cc] Error 2

cliffordwolf, 2018-05-14:
(1) Use git head, not the yosys-0.7 release from 2016. (ABC has moved from bitbucket to github.)
(2) `hg: not found` tells me that you ignored the install instructions that said to install `hg`.

moohtuh, 2018-05-14:
Sorry to be posting again; I used the git head now getting this meesage(again it builds, but after coming to the abc stage same result):

muhtadic16@ubuntu:~/Downloads/yosys-master$ make
[  0%] Building abc/abc-f23ea8e
Pulling ABC from https://github.com/berkeley-abc/abc:
+ test -d abc
+ git clone https://github.com/berkeley-abc/abc abc
/bin/sh: 4: git: not found
+ cd abc
/bin/sh: 5: cd: can't cd to abc
Makefile:459: recipe for target 'abc/abc-f23ea8e' failed
make: *** [abc/abc-f23ea8e] Error 2

moohtuh, 2018-05-15:
Please help.

cliffordwolf, 2018-05-15:
The readme tells you to install git:

[image]

The error message tells you it can't find git:

[image]

Install git!

moohtuh, 2018-05-15:
Thanks very much! Now I am getting this error- all previous tests before mem2reg passed.

Test: macros -> ok
Test: mem2reg -> ERROR!
Note: Make sure that 'iverilog' is an up-to-date git checkout of Icarus Verilog.
../tools/autotest.mk:10: recipe for target 'mem2reg.v' failed
make[1]: *** [mem2reg.v] Error 1
make[1]: Leaving directory '/home/muhtadic16/Downloads/yosys-master/tests/simple'
Makefile:491: recipe for target 'test' failed
make: *** [test] Error 2

This was in the mem2reg.err file -

Warnings: 1 unique messages, 1 total
End of script. Logfile hash: e04b7ffa5a
CPU: user 0.01s system 0.00s, MEM: 40.28 MB total, 7.45 MB resident
Yosys 0.7+ (git sha1 UNKNOWN, clang  -fPIC -Os)
Time spent: 99% 1x read_verilog (0 sec), 0% 1x test_autotb (0 sec)
+ false
+ compile_and_run mem2reg_tb_ref mem2reg_out_ref mem2reg_tb.v mem2reg_ref.v
+ exe=mem2reg_tb_ref
+ output=mem2reg_out_ref
+ shift 2
+ false
+ false
+ iverilog '-Doutfile="mem2reg_out_ref"' -s testbench -o mem2reg_tb_ref mem2reg_tb.v mem2reg_ref.v
mem2reg_ref.v:76: sorry: constant user functions are not currently supported: depth2Index().
mem2reg_ref.v:76: error: array intermediate index must be a constant in this context.
mem2reg_ref.v:77: sorry: constant user functions are not currently supported: depth2Index().
mem2reg_ref.v:77: error: array intermediate index must be a constant in this context.
2 error(s) during elaboration.

Please let me know if I am doing anything wrong; apologies for repeated messages.

tomverbeure, 2018-05-16:
Rule of thumb: when something fails, start reading the error messages from top to bottom and fix things in that order.

Check out the *third* line:

> Note: Make sure that 'iverilog' is an up-to-date git checkout of Icarus Verilog.

You're using iverilog from an Ubuntu package, version 0.9.7. 

Is an Ubuntu package an up-to-date git checkout?

No. It is not.

Solution to your problem: "Make sure that 'iverilog' is an up-to-date git checkout of Icarus Verilog."

If things still fail after that, repeat the process: read the error messages. Read them again. Fix what can be fixed.

> Please let me know if I am doing anything wrong; apologies for repeated messages.

What you're doing wrong is not reading the error messages.

moohtuh, 2018-05-16:
Now I have built the whole Yosys and successfully all the tests are passed and I get this message in the terminal at the end: 

............
Running initval.ys..
Warning: ignoring initial value on non-register: \bar [1:0]
Running share.ys..
Running sizebits.ys..
Running splice.ys..

  Passed "make test".

But when I give this command:           yosys > read_verilog  -dump_ast1  aes_kexp128.v

I get :           The program 'yosys' is currently not installed. You can install it by typing:
sudo apt install yosys

Even when I give the command :   yosys        in terminal 
It says:               The program 'yosys' is currently not installed. You can install it by typing:
sudo apt install yosys
 
All I need is the AST formats and the write_verilog i.e. getting back to source code from AST.

I don't know where I did wrong ...no errors anywhere this time ...Please help. I feel bad for repeatedly bothering you guys.