YosysHQ/yosys #552: cmake support

nathan-prat, 2018-05-21:
Hello,

Do you plan to add CMake support? 
This would also allow to build out-of-source.
It would also help integrate yosys as a library in CMake based projects(ie libyosys.so).

I have a working(still WIP but functional CMakeLists.txt, see attached): would you be interested in a pull request(this would be linux only)?

[CMakeLists.txt](https://github.com/YosysHQ/yosys/files/2022244/CMakeLists.txt)

PS: I don't know if opening an issue is the best way to create a "feature request", sorry if it's not the case.

cliffordwolf, 2018-05-21:
> Do you plan to add CMake support?

No. And I'm actually pretty happy with our Makefile-based build system. I would not change it unless there are (1) really good arguments for changing it and (2) a binding long-term commitment from someone to maintain the cmake based build system.

> This would also allow to build out-of-source.

There already is support for building Yosys out-of-source. (But not sure how well it is tested right now tbh.)

Simply `cd` to your (empty) build dir and run `make -f <yosys-src-dir>/Makefile` there.

closed 2018-05-21 by cliffordwolf

tomager, 2019-12-16:
I know it's late, but I just wanted to chime in and say cmake support would be awesome. Maybe it could exist in parallel to the current build system and maintained strictly by those interested, but kept in upstream repo for ease of use.

** The big advantage of cmake is that it makes yosys portable to non-Unix systems or more complex build environments without any or much extra work. For example targeting Ninja, native XCode, native MSVC, WASM, MXE, clang, many cross compiled environments, etc, is very simple out of the box without direct support from yosys. Also many niceties are handled automatically/properly such as header dependency tracking, inter-library dependency tracking, proper shared/static library handling, out of source builds, separated debug/release builds on windows, soon C++ modules support, etc.

Build rules in cmake are written in an "intentional" way that is somewhat abstracted away from the particular tools/system, but is still easy to read and maintain. IMO much easier to maintain than traditional makefiles once you get over the fairly small learning curve.

** CMake is extremely popular these days. A large fraction of all C/C++ projects on github use it, including many major projects such as LLVM. It's the most popular build system in 2019 [1], and I suspect for new projects is much more dominant. Including yosys into another cmake project is trivial if yosys itself is using cmake.

[1] https://www.jetbrains.com/lp/devecosystem-2019/cpp/