YosysHQ/yosys #553: just a normal query

moohtuh, 2018-05-21:
May I ask if write_verilog produces verilog code from the RTLIL representation? If so, how do I input the RTLIL representation for a specific design?

I looked in the manual and tried to get the verilog code for a simple verilog input code : 

module top
  (
   input CLK,
   input RST,
   input enable,
   input [31:0] value,
   output [7:0] led
  );
  reg [31:0] count;
  reg [7:0] state;
  assign led = count[23:16];
  always @(posedge CLK) begin
    if(RST) begin                            
      count <= 0;
      state <= 0;
    end else begin
      if(state == 0) begin
        if(enable) state <= 1;
      end else if(state == 1) begin
        state <= 2;
      end else if(state == 2) begin
        count <= count + value;
        state <= 0;
      end
    end
  end

But the output is jargon; I don't understand anything. Please help.

cliffordwolf, 2018-05-22:
> May I ask if write_verilog produces verilog code from the RTLIL representation?

Yes it does.

> If so, how do I input the RTLIL representation for a specific design?

What?

> But the output is jargon; I don't understand anything. Please help.

You don't quote the output. You don't say what commands you are running. All you say is that you do some undisclosed thing with this design and that you do not understand the output.

Are you serious? What exactly is your expectation? That someone will magically read your mind? How do you expect anyone to help you if you do not even provide sufficient information to reproduce whatever you are seeing, nor say what it is that you are seeing?

ZipCPU, 2018-05-22:
I thought I might try this myself.  You can see the results and the script I used [here](http://zipcpu.com/tmp/testv.tgz).

Looking over the output file, I find that it does indeed appear to use the set of elements given.  It's not "jargon"--I think the original poster didn't understand the meaning of that word.  I think the OP was expecting something else, perhaps something he might turn into his instructor rather than what it is--the computer output of synthesized verilog code.  As computer output, it shows how the yosys program assigned various logic stages to the "hardware" elements it was given and required to create logic from.  This is what I would expect of the output.  I see nothing unexpected there.

I do think it's rather strange that the OP would use an 8-bit state variable to record 3 states.  Looks like yosys caught that and trimmed it down nicely as well.

I'm going to recommend closing this issue.

closed 2018-05-22 by cliffordwolf

AlexDaniel, 2018-05-22:
> I would ask the OP to open a discussion on IRC

By the way, it's also possible to do that on [stackoverflow](https://stackoverflow.com/questions/tagged/yosys).

> Are you serious?

IMO that was unnecessarily aggressive.

<br>
Have a nice day everyone!

moohtuh, 2018-05-22:
Please accept my apology everyone; I actually do understand it was my fault I wanted help without providing you guys the pertinent information you needed.

#1.         I saw ZipCPU changed the alignment and some names of the source code like : 

module top
  (
   input CLK,
   input RST,
   input enable,
   input [31:0] value,
   output [7:0] led
  );

into
 
module top(input wire i_clk, input wire i_reset, input wire i_enable,
	input wire [31:0] i_value, output wire [7:0] o_led);

So basically adding i as a prefix for inputs and o for outputs; are these necessary? If so which section of the manual covers this if I may ask?

#2.  I get this output if I don't give the synth command in between, that is : 
read_verilog code.v
write_verilog code.v

(* src = "new.v:1" *)
module top(i_clk, i_reset, i_enable, i_value, o_led);
  (* src = "new.v:9" *)
  reg [31:0] _0_;
  (* src = "new.v:9" *)
  reg [7:0] _1_;
  (* src = "new.v:24" *)
  wire [31:0] _2_;
  (* src = "new.v:16" *)
  wire _3_;
  (* src = "new.v:20" *)
  wire _4_;
  (* src = "new.v:22" *)
  wire _5_;
  (* src = "new.v:4" *)
  reg [31:0] count;
  (* src = "new.v:1" *)
  input i_clk;
  (* src = "new.v:1" *)
  input i_enable;
  (* src = "new.v:1" *)
  input i_reset;
  (* src = "new.v:2" *)
  input [31:0] i_value;
  (* src = "new.v:2" *)
  output [7:0] o_led;
  (* src = "new.v:5" *)
  reg [7:0] state;
  assign _2_ = count + (* src = "new.v:24" *) i_value;
  assign _3_ = state == (* src = "new.v:16" *) 32'd0;
  assign _4_ = state == (* src = "new.v:20" *) 32'd1;
  assign _5_ = state == (* src = "new.v:22" *) 32'd2;
  always @* begin
    _0_ = count;
    _1_ = state;
    casez (i_reset)
      1'h1:
        begin
          _0_ = 32'd0;
          _1_ = 8'h00;
        end
      default:
          casez (_3_)
            1'h1:
                casez (i_enable)
                  1'h1:
                      _1_ = 8'h01;
                  default:
                      /* empty */;
                endcase
            default:
                casez (_4_)
                  1'h1:
                      _1_ = 8'h02;
                  default:
                      casez (_5_)
                        1'h1:
                          begin
                            _0_ = _2_;
                            _1_ = 8'h00;
                          end
                        default:
                            /* empty */;
                      endcase
                endcase
          endcase
    endcase
  end
  always @(posedge i_clk) begin
      count <= _0_;
      state <= _1_;
  end
  assign o_led = count[23:16];
endmodule

May I ask if yosys is trying to get back to the source code here? And if so is it intended to go there with the given set of commands or write_verilog is only written for post-synthesis?

Apologies again.

ZipCPU, 2018-05-22:
The differences in identifier names is my own personal preference.  I made the change because it helped me understand more about what your code was doing.  Such prefixes are not required.

I think what you are misunderstanding is that yosys is *not* attempting to return the original source code.  Instead, yosys has transformed your original source code by mapping your code onto particular hardware elements that exist in the system it has been synthesized for.  The verilog that is returned is a module that only uses those source code elements.  For example, had you synthesized for an iCE40, the design would now reference each iCE40 element and its configuration.  No, it isn't all that legible.  If you wanted legible, you'd stick with what you started with.  Instead, the names and logic reflect the internal synthesis structures--identifiers are generated internally and designed for computer understanding, not human understanding.  If you want the original source file, relax, you've still got it.

moohtuh, 2018-05-22:
May I ask if I have the option of changing the particular hardware element Yosys maps/synthesizes my code for?

ZipCPU, 2018-05-22:
Sure!  But expect a different output file if you do so.  The Verilog output is very dependent upon which hardware type you choose.

Dan

moohtuh, 2018-05-22:
Two more questions if I may : 

So write_verilog transfers the intermediate RTLIL information of a source code to synthesizing according to the platform intended.

But suppose I wanted to see the RTLIL step for the code, capture it and edit some of it and then synthesize it, would it be possible with this tool?

I suppose I can see the RTLIL of any code with the dump_ast1 or dump_ast2 but can I make any changes to them before the synthesize step?

Also, which AST format: ast1 or ast2 or ILANG is the most strongest i.e. doesn't lose any information? 

Thanks for all the answers, everyone. Much obliged.

moohtuh, 2018-05-23:
Anybody, please ?

ZipCPU, 2018-05-23:
This is the wrong forum for such questions.  This is the forum for reporting if you find something broken with yosys.  See above for other some other forums where you might ask your question: the #yosys channel on [IRC](https://webchat.freenode.net/) (be prepared to wait 24hrs or more for your answer), [stackoverflow](https://stackoverflow.com/questions/tagged/yosys), or [reddit](https://www.reddit.com/r/yosys/) (answers sometimes take a week or so).

moohtuh, 2018-05-29:
@ZipCPU 

Hi ZipCPU,

Could you tell me how to synthesize a source code for different platforms? In the manual, it's mentioned we can convert source code into different netlists according to different FPGAs like Xilinx, Gowing, etc.

Is there any specific flag for that? Please help.