YosysHQ/yosys #554: RFE: Support for command line or synthesis script `defines

tomverbeure, 2018-05-28:
Say I have an existing code base that has constructs like this:
```
`ifndef SYNTHESIS
    $display("Printer port on fire");
`endif
```
I can easily filter out these constructs during synthesis by specifying -DSYNTHESIS on the command line or somewhere as a global variable in a synthesis script.

I haven't found a solution on how to do this with Yosys.

I've seen the following reddit comment about "Resolve `define"
https://www.reddit.com/r/yosys/comments/56ac1y/resolve_define/
which states "The right way to do this is by using includes."

That doesn't make me very hopeful. The proposed solution requires either wrapping *all* files where this construct is used or modifying all these files with an `include. That's not a happy prospect to work around a missing feature that's supported by all other synthesis and simulation tools that I know of.

Am I overlooking some trivial way of accomplishing this that doesn't require the reddit solution? If not, I'd really like to vote to have these kind of global `define to be added as a feature for a future version of Yosys.

Tom

cliffordwolf, 2018-05-28:
(1) The Verilog synthesis standard says a synthesis tool should always define `SYNTHESIS`. Yosys does that, so there is nothing you need to do.

(2) The `read_verilog` command accepts `-D foo` or `-D foo=bar` (the space is optional). This sets a define only for the verilog files specified in this one invocation of `read_verilog`.

(3) The command `verilog_defines -D foo=bar` sets the specified define for all following `read_verilog` calls.

(3) That reddit post is from 2016. Current Yosys preserves defines between calls to `read_verilog`.

closed 2018-05-28 by cliffordwolf

cliffordwolf, 2018-05-28:
No worries.