YosysHQ/yosys #555: Some queries

moohtuh, 2018-05-29:
I had some questions related to how Yosys works; I asked them in the webchat but no one responds.

Please help me, someone.

1.  For any source code, do we get AST2 from AST1?

2. For any source code, if I want to make changes in the AST slightly and recreate the netlist using write_verilog (which as I understand, converts the AST to platform-specific Verilog netlists), which module of the Yosys do I have to make the changes in?

**Example: Suppose for this source code:** 

`module top
  (
   input CLK,
   input RST,
   input enable,
   input [31:0] value,
   output [7:0] led
  );
  reg [31:0] count;
  reg [7:0] state;
  assign led = count[23:16];
  always @(posedge CLK) begin
    if(RST) begin                            
      count <= 0;
      state <= 0;
    end else begin
      if(state == 0) begin
        if(enable) state <= 1;
      end else if(state == 1) begin
        state <= 2;
      end else if(state == 2) begin
        count <= count + value;
        state <= 0;
      end
    end
  end
endmodule`

**I get this AST1 :** 

`    AST_MODULE <new.v:1> [0x17a16a0] str='\top'
      AST_WIRE <new.v:3> [0x179d640] str='\CLK' input port=1
      AST_WIRE <new.v:4> [0x179d740] str='\RST' input port=2
      AST_WIRE <new.v:5> [0x179d890] str='\enable' input port=3
      AST_WIRE <new.v:6> [0x179d9c0] str='\value' input port=4
        AST_RANGE <new.v:6> [0x179dd70]
          AST_CONSTANT <new.v:6> [0x179db10] bits='00000000000000000000000000011111'(32) signed range=[31:0] int=31
          AST_CONSTANT <new.v:6> [0x179dc40] bits='00000000000000000000000000000000'(32) signed range=[31:0]
      AST_WIRE <new.v:7> [0x179de70] str='\led' output port=5
        AST_RANGE <new.v:7> [0x179e210]
          AST_CONSTANT <new.v:7> [0x179dfb0] bits='00000000000000000000000000000111'(32) signed range=[31:0] int=7
          AST_CONSTANT <new.v:7> [0x179e0e0] bits='00000000000000000000000000000000'(32) signed range=[31:0]
      AST_WIRE <new.v:9> [0x179e780] str='\count' reg
        AST_RANGE <new.v:9> [0x179e880]
          AST_CONSTANT <new.v:9> [0x179e980] bits='00000000000000000000000000011111'(32) signed range=[31:0] int=31
          AST_CONSTANT <new.v:9> [0x179ea80] bits='00000000000000000000000000000000'(32) signed range=[31:0]
      AST_WIRE <new.v:10> [0x179ecf0] str='\state' reg
        AST_RANGE <new.v:10> [0x179edf0]
          AST_CONSTANT <new.v:10> [0x179eef0] bits='00000000000000000000000000000111'(32) signed range=[31:0] int=7
          AST_CONSTANT <new.v:10> [0x179eff0] bits='00000000000000000000000000000000'(32) signed range=[31:0]
      AST_ASSIGN <new.v:11> [0x179f240]
        AST_IDENTIFIER <new.v:11> [0x179e310] str='\led'
        AST_IDENTIFIER <new.v:11> [0x179f140] str='\count'
          AST_RANGE <new.v:11> [0x179ebf0]
            AST_CONSTANT <new.v:11> [0x179e680] bits='00000000000000000000000000010111'(32) signed range=[31:0] int=23
            AST_CONSTANT <new.v:11> [0x179e450] bits='00000000000000000000000000010000'(32) signed range=[31:0] int=16
      AST_ALWAYS <new.v:12> [0x179f340]
        AST_POSEDGE <new.v:12> [0x179f5d0]
          AST_IDENTIFIER <new.v:12> [0x179f4d0] str='\CLK'
        AST_BLOCK <new.v:12> [0x179f6d0]
          AST_BLOCK <new.v:13> [0x179f7f0]
            AST_CASE <new.v:13> [0x179f9f0]
              AST_REDUCE_BOOL <new.v:13> [0x179fe50]
                AST_IDENTIFIER <new.v:13> [0x179f8f0] str='\RST'
              AST_COND <new.v:13> [0x179fbf0]
                AST_CONSTANT <new.v:13> [0x179fcf0] bits='1'(1) range=[0:0] int=1
                AST_BLOCK <new.v:13> [0x179faf0]
                  AST_BLOCK <new.v:14> [0x179ffb0]
                    AST_ASSIGN_LE <new.v:14> [0x17a02f0]
                      AST_IDENTIFIER <new.v:14> [0x17a00b0] str='\count'
                      AST_CONSTANT <new.v:14> [0x17a01f0] bits='00000000000000000000000000000000'(32) signed range=[31:0]
                    AST_ASSIGN_LE <new.v:15> [0x17a0660]
                      AST_IDENTIFIER <new.v:15> [0x17a03f0] str='\state'
                      AST_CONSTANT <new.v:15> [0x17a0530] bits='00000000000000000000000000000000'(32) signed range=[31:0]
              AST_COND <new.v:16> [0x17a0860]
                AST_DEFAULT <new.v:16> [0x17a0960]
                AST_BLOCK <new.v:16> [0x17a0760]
                  AST_BLOCK <new.v:17> [0x17a0a80]
                    AST_CASE <new.v:17> [0x17a0f40]
                      AST_REDUCE_BOOL <new.v:17> [0x17a1380]
                        AST_EQ <new.v:17> [0x17a0e40]
                          AST_IDENTIFIER <new.v:17> [0x17a0b80] str='\state'
                          AST_CONSTANT <new.v:17> [0x17a0d10] bits='00000000000000000000000000000000'(32) signed range=[31:0]
                      AST_COND <new.v:17> [0x17a1140]
                        AST_CONSTANT <new.v:17> [0x17a1240] bits='1'(1) range=[0:0] int=1
                        AST_BLOCK <new.v:17> [0x17a1040]
                          AST_BLOCK <new.v:18> [0x17a5ec0]
                            AST_CASE <new.v:18> [0x17a60c0]
                              AST_REDUCE_BOOL <new.v:18> [0x17a64c0]
                                AST_IDENTIFIER <new.v:18> [0x17a5fc0] str='\enable'
                              AST_COND <new.v:18> [0x17a62c0]
                                AST_CONSTANT <new.v:18> [0x17a63c0] bits='1'(1) range=[0:0] int=1
                                AST_BLOCK <new.v:18> [0x17a61c0]
                                  AST_ASSIGN_LE <new.v:18> [0x17a67f0]
                                    AST_IDENTIFIER <new.v:18> [0x17a65c0] str='\state'
                                    AST_CONSTANT <new.v:18> [0x17a66c0] bits='00000000000000000000000000000001'(32) signed range=[31:0] int=1
                      AST_COND <new.v:19> [0x17a69f0]
                        AST_DEFAULT <new.v:19> [0x17a6af0]
                        AST_BLOCK <new.v:19> [0x17a68f0]
                          AST_CASE <new.v:19> [0x17a6f90]
                            AST_REDUCE_BOOL <new.v:19> [0x17a73d0]
                              AST_EQ <new.v:19> [0x17a6e90]
                                AST_IDENTIFIER <new.v:19> [0x17a6c30] str='\state'
                                AST_CONSTANT <new.v:19> [0x17a6d60] bits='00000000000000000000000000000001'(32) signed range=[31:0] int=1
                            AST_COND <new.v:19> [0x17a7190]
                              AST_CONSTANT <new.v:19> [0x17a7290] bits='1'(1) range=[0:0] int=1
                              AST_BLOCK <new.v:19> [0x17a7090]
                                AST_BLOCK <new.v:20> [0x17a7530]
                                  AST_ASSIGN_LE <new.v:20> [0x17a78c0]
                                    AST_IDENTIFIER <new.v:20> [0x17a7630] str='\state'
                                    AST_CONSTANT <new.v:20> [0x17a7790] bits='00000000000000000000000000000010'(32) signed range=[31:0] int=2
                            AST_COND <new.v:21> [0x17a7ac0]
                              AST_DEFAULT <new.v:21> [0x17a7bc0]
                              AST_BLOCK <new.v:21> [0x17a79c0]
                                AST_CASE <new.v:21> [0x17a8060]
                                  AST_REDUCE_BOOL <new.v:21> [0x17a8480]
                                    AST_EQ <new.v:21> [0x17a7f60]
                                      AST_IDENTIFIER <new.v:21> [0x17a7ce0] str='\state'
                                      AST_CONSTANT <new.v:21> [0x17a7e30] bits='00000000000000000000000000000010'(32) signed range=[31:0] int=2
                                  AST_COND <new.v:21> [0x17a8260]
                                    AST_CONSTANT <new.v:21> [0x17a8360] bits='1'(1) range=[0:0] int=1
                                    AST_BLOCK <new.v:21> [0x17a8160]
                                      AST_BLOCK <new.v:22> [0x17a85e0]
                                        AST_ASSIGN_LE <new.v:22> [0x17a8b20]
                                          AST_IDENTIFIER <new.v:22> [0x17a86e0] str='\count'
                                          AST_ADD <new.v:22> [0x17a89e0]
                                            AST_IDENTIFIER <new.v:22> [0x17a87e0] str='\count'
                                            AST_IDENTIFIER <new.v:22> [0x17a88e0] str='\value'
                                        AST_ASSIGN_LE <new.v:23> [0x17a8eb0]
                                          AST_IDENTIFIER <new.v:23> [0x17a8c40] str='\state'
                                          AST_CONSTANT <new.v:23> [0x17a8d80] bits='00000000000000000000000000000000'(32) signed range=[31:0]
`

Now suppose I want to make changes in the AST- like changing the range of str 'led' to some other values like from [7:0] to [8:0]. And then, may be adding some lines in the AST to add some more logic.

Then using that modified AST, I would get new RTLIL and using write_verilog new netlists.

Which module do I need to change in Yosys for that ? Please help.

cliffordwolf, 2018-05-30:
1. The manual describes what AST is used for, including that it only exist internal to `read_verilog`. Read the manual, read the code, none of your "questions" make any sense. `write_verilog` never sees the AST.

2. You gave [this comment](https://github.com/YosysHQ/yosys/issues/553#issuecomment-391486887) a thumbs up. Yet you completely ignore what it said. Maybe you shouldn't.

closed 2018-05-30 by cliffordwolf

ZipCPU, 2018-05-30:
Github issues are where bugs should be reported, not where discussions should take place of this nature.  Please ask your question on either IRC, Reddit, or stack overflow.

As for your conclusion that no one responds on webchat, perhaps you didn't wait long enough.  Specifically, I can read your last question [here](https://irclog.whitequark.org/yosys/2018-05-23), and see the response [here](https://irclog.whitequark.org/yosys/2018-05-24).