YosysHQ/yosys #556: Register changes values without assignment

ZipCPU, 2018-05-31:
## Steps to reproduce the issue

See the example at [zipcpu.com/tmp/mem-bug.tgz](http://zipcpu.com/tmp/rom-bug.tgz).  In that tar file is a read only memory design, and a data register that is assumed to match this memory at some arbitrary address.  Neither are values are ever set during the design, save that the register is assumed to have a value at time zero.  I anticipate adding a $readmemh file to this design later.

There's also a very basic SymbiYosys script file in the same tar file.

When the SymbiYosys script runs, it creates a failing trace that requires the `f_data` value to change--a register that is *never set at all* during the design.

## Expected behavior

I would expect a register that is never assigned to would maintain its initial (assumed) value.  `f_data` should be such a register.  It's initial value is assumed, and it is never afterwards assigned to.

## Actual behavior

The register changes value on the first clock edge.

## Annoying issue behavior

This issue tracker doesn't let me upload the .v and .sby files necessary to reproduce this issue here, complaining that the file type isn't supported.  :/  For this reason, the files are placed on [zipcpu.com](http://zipcpu.com)

cliffordwolf, 2018-06-01:
> it creates a failing trace that requires the f_data value to change--a register that is never set at all during the design.

Since it is never assigned anything it has the value `32'h XXXX_XXXX`. This is effectively converted to an "anyseq" driver because there is no guarantee in Verilog that a constant `X` simulation value will be sampled as constant 0 or 1 in an implementation. So this is the intended behavior.

I've now added a `-anyconst` mode to `setundef` that allows you to use "anyconst" semantics instead. (Requires f273291 or newer.) Simply add the following command to the end of your .sby `[script]` section:

    setundef -undriven -anyconst

> This issue tracker doesn't let me upload the .v and .sby files

Yes. It's strange.. But jfyi, you can upload .tar.gz files and .zip files. (But the .tgz file extension isn't supported.)

closed 2018-06-01 by cliffordwolf

ZipCPU, 2018-06-01:
Let me suggest that $anyconst should be the default value for a register that is never assigned, rather than $anyseq.  Think of a register within hardware--if never assigned, it's not going to be changing values over time.

cliffordwolf, 2018-06-02:
That would create a formal model that is narrower that the envelop of allowed behaviors for values that are undefined.

According to the verilog spec that reg has a constant value that does not change, and that constant value is `X`. The register is not changing values! Instead, it is non-deterministic if sampling that `X` in a boolean context will return 0 or 1 *for each individual sample taken*.

(This is still narrower than complete proper X-propagation, as is implemented in the Yosys `sat` command, but is currently not available in SymbiYosys. But I think we should not deliberately make the situation worse by narrowing it even further.)