YosysHQ/yosys #557: conditional initial value for async register

edcote, 2018-06-01:
RE: https://github.com/freechipsproject/rocket-chip/commit/41702a950f2fdc9a247b654869bff6a9389a9bb2

A statement was added in async reg initial block that is unsupported by yosys.

## Steps to reproduce the issue

[racecond.zip](https://github.com/YosysHQ/yosys/files/2064330/racecond.zip)

1. unzip racecond.zip
1. yosys -s ys

## Expected behavior

No error.

## Actual behavior

~~~~
3.2.3. Executing PROC_INIT pass (extract init attributes).
Found init rule in `\AsyncResetReg.$proc$AsyncResetReg.v:11$3'.
ERROR: Failed to get a constant init value for \q_reg: $2\q_reg[0:0]
~~~~

## Location of failure

~~~~
module AsyncResetReg (
                      input      d,
                      output     q,
                      input      en,

                      input      clk,
                      input      rst);

   reg                           q_reg;

   initial begin
// ---8<---
      if (rst) begin
        q_reg = 1'b0;
      end
// --->8---
   end

   always @(posedge clk or posedge rst) begin
      if (rst) begin
         q_reg <= 1'b0;
      end else if (en) begin
         q_reg <= d;
      end
   end

   assign q = rst ? 1'b0 :  q_reg;

endmodule // AsyncResetReg
~~~~

cliffordwolf, 2018-06-02:
As discussed in freechipsproject/rocket-chip@41702a9, this is non-synthesisable code.

closed 2018-06-02 by cliffordwolf

edcote, 2018-06-03:
Thank you, Cliff.