YosysHQ/yosys #574: Possible Miscompile during Block RAM Inference?

cr1901, 2018-06-29:
## Steps to reproduce the issue
To test something regarding BRAM inference, I created a `yosys` script that duplicates the commands of `synth_ice40` _up until the `fine`_ label, at which point I write out to a file.

I compile a verilog module called `bram.v` (a generic inferable block RAM) using two scripts- `bram.ys`, which uses the `synth_ice40` command directly, and `bram-man.ys` which is described in the previous paragraph. Just out of curiosity, I also provide an `extra` target in the Makefile, which finishes up synthesis using `synth_ice40 -run fine:`. Here is a test case: [bram.zip](https://github.com/YosysHQ/yosys/files/2147669/bram.zip)

## Expected behavior
Since the commands being run in each script are nominally the same I would have expected both scripts to have similar output, with inferred BRAM mapped to `SB_RAM40_4K` in both files.

## Actual behavior
Instead, the synthesis script `bram-man.ys` almost matches the input `bram.v` file, as if most (but not all) optimizations and passes were ignored on the design; write enables got lifted out of the `always` block, however. As expected, if you run `make extra`, yosys ultimately maps memory to a rather large amount of flip-flops instead.

Is this a bug, or does `synth_ice40` do more than provide a thin wrapper around the listed commands in the manual on page 134-135?

daveshah1, 2018-06-29:
I think this line in bram-man.ys

`memory_bram rules +/ice40/brams.txt`

should be

`memory_bram -rules +/ice40/brams.txt`

Otherwise you are running `memory_bram` only on non-existent module `rules`.

cr1901, 2018-06-30:
... ... ...

I withdraw my issue...

closed 2018-06-30 by cr1901