YosysHQ/yosys #577: Specifying -viewer with show if -format is also not specified

mithro, 2018-07-07:
## Steps to reproduce the issue

Specifying `-viewer` but not `-format` doesn't work.

 * `show` works
 * `show -viewer xdot` fails with xdot saying 'No such file ~/.yosys_show.svg'
 * `show -viewer xdot -format dot` works
 * `show -viewer eog -format svg` works
 * `show -viewer xdot -format svg` doesn't work but is user error, so not yosys's fault (xdot can't read an svg file).

Yosys version: `Yosys 0.7+578 (git sha1 a7281930, x86_64-conda_cos6-linux-gnu-gcc 7.2.0 -fPIC -Os)`

## Expected behavior

One of the following;
 * It generates the svg file.
 * It passes the `.dot` file argument to the viewer command.

mithro, 2018-07-07:
Looks like this has already been fixed in the latest yosys version.`Yosys 0.7+612 (git sha1 8b92ddb9, gcc 7.3.0-5 -fPIC -Os)`

closed 2018-07-07 by mithro

slagernate, 2023-05-23:
`show -viewer xdot` is still failing with `no such file or directory /home/user/.yosys_show.svg`

Maybe a separate issue, and maybe too late, but it seems `show` by itself should default to opening the viewer as well (unless, say, `-quiet` / `-q` is specified).