YosysHQ/yosys #579: ERROR: Assert `chunk.width >= 0' failed in kernel/rtlil.cc:3264.

sliedes, 2018-07-10:
## Steps to reproduce the issue

On current HEAD of yosys (commit 8b92ddb9), running `yosys -S bug.v` causes an assertion failure:

```
$ ../deps/yosys/yosys -S bug.v
[...]
 Yosys 0.7+612 (git sha1 8b92ddb9, clang 6.0.0 -fPIC -Os)

-- Parsing `bug.v' using frontend `verilog' --

1. Executing Verilog-2005 frontend.
Parsing Verilog input from `bug.v' to AST representation.
Generating RTLIL representation for module `\bug'.
ERROR: Assert `chunk.width >= 0' failed in kernel/rtlil.cc:3264.
```

Here is a minimized small bug.v:

```
module bug(input [0:3] x, input [0:32] y, output [0:2] out);
    assign out = y[x:x+2];
endmodule
```

tomverbeure, 2018-10-03:
@cliffordwolf 

Right now, Yosys asserts on the example code above.
It doesn't assert on descending ranges. (e.g. `y[x+2:x]`)

This is expected as indicated in the [source code comments](https://github.com/YosysHQ/yosys/blob/76baae4b946cdeb04026120b495c87a6146358d0/frontends/ast/genrtlil.cc#L945-L947).

However, from my reading of the LRM (1800-2018, section 11.5.1, page 280), this is not allowed:

[image]

Both iVerilog and verilator croak on both `[x:x+2]` and `[x+2:x]`.

This bug report can be fixed as follows:

1. print an error message for the [x:x+2] case instead of an assert.
1. print an error message for the [x:x+2] case _and_ for the [x+2:x] case. (That is: a regression in functionality, but adhering to the spec.)
1. make the [x:x+2] case work.

I assume that 1 is the preferred option?

daveshah1, 2019-05-18:
No longer seeing the assertion failure with this testcase on latest master.

closed 2019-05-18 by daveshah1