YosysHQ/yosys #582: Error failed to detect width for identifier

adamdai, 2018-07-16:
I am trying to compile the verilog file `stencil.v`
[stencil.v.zip](https://github.com/YosysHQ/yosys/files/2199414/stencil.v.zip)
with the following yosys command
```
yosys -q -p 'synth_ice40 -top main -blif stencil.blif' stencil.v
```
and receive the following error
```
ERROR: Failed to detect width for identifier \valid at stencil.v:1183!
make: *** [stencil.bin] Error 1
```
Upon initial inspection it seems like the verilog is fine, so I am wondering if there might be a bug in yosys that is causing this error. 

@leonardt

udif, 2018-07-16:
On your design, 'valid' is both an output *(line 1039) and an instance name (line 1143).
I don't know if this is legal Verilog, but I wouldn't feel comfortable with it.

leonardt, 2018-07-16:
@udif ah good catch, makes sense. That should be an easy fix on our side.

closed 2018-07-16 by adamdai