YosysHQ/yosys #598: Feature request: Splitting and generating hierarchical designs for larger designs

thesourcerer8, 2018-08-05:
I would like to have a functionality in Yosys so that I can define certain technology boundaries (the number of logic blocks in an FPGA, the number of cells in an ASIC design), and when Yosys detects that the given Verilog design cannot be implemented within those boundaries, that it splits up (perhaps with a special command, only on demand) the design into 2 or more individual sub-designs, and then synthesizes the individual sub-designs seperately, and outputs the whole hierarchical design at the end too. So for example for an FPGA design it would tell you that you need 3 of the chosen FPGAs, and it would output the respective design files for each of those 3 FPGAs, and also a netlist (or something similar) that communicates the top-level connectivity between all 3 FPGAs, which could then be used to generate a PCB board design. The reason why I want to have this implemented in Yosys, and not somewhere else in the toolchain before or afterwards is because I want to be able to process larger Verilog designs, and I think that it is an optimization problem to seperate a design into several designs, given the usually limited number of possible signal traces/tracks/pins between the seperate FPGA's / cores, and perhaps also the timing constraints due to longer connections between different sub-designs. The workaround without this functionality is to manually split the Verilog design into smaller parts, which seems to be feasible for something like 2-3 sub-designs, but I doubt that it is a good way for >10 sub-designs. Splitting the design into sub-designs in later stages is not such a good ideas I think, because Yosys understands the requirements of the design much better than a tool at a later stage, I think.

jeremyherbert, 2018-08-05:
As far as my expertise, I'm no clifford, but I think that this is basically an intractable problem for an arbitrary circuit. The issue is that connecting two FPGAs (or more) together is simple in theory, but is impractical to do automatically in practice. Just a netlist is not sufficient to connect them; what about trace length, trace impedance, etc? How do you sync clocks between the devices? how do you do timing simulations without knowing the routing and impedances on the interconnect? What about the inductance in the packaging/bonding itself? etc etc etc... 

This is all notwithstanding that yosys is not a PAR tool (or at least last time I checked), and I believe as such cannot do any optimisation based on propagation delay which would be necessary in a situation like this.

If you need to split your design into parts, you would need to think long and hard about how you shared data between the two (or more) parts such that it can occur at a lower clock rate and between two different clock domains in a way that didn't ruin your performance. You might need to use some vendor-specific SerDes units, and write bunch of code to wrap around these. I don't think there are any HDLs out there which are expressive enough to allow the designer to specify these sort of constraints in a generic fashion.

thesourcerer8, 2018-08-05:
The idea is that it might be possible to split it into parts, where the interconnect between the parts is less demanding than the connections inside the parts. I don't expect guarantees that everything will be fine, but I would like to be able to get the best possible splitting. I guess that the biggest restriction will be actually the number of interconnects, not the speed.
At the moment, I either have to split the design manually before I shovel it into yosys, or I have to split the resulting netlist afterwards randomly. If the manual splitting becomes infeasible, I would hope that yosys can improve the situation of splitting the netlist randomly. I am not aiming for a perfectly split high-speed design, but I believe that something better than randomly splitting netlists should be possible.