YosysHQ/yosys #601: Port ordering in JSON output

tilk, 2018-08-06:
Is there any way to recover the ordering of ports from the JSON output? I'm working on visualizing the netlists generated by Yosys, and using the original order in the visualization would help clarity. It seems that the ports are output in JSON in the right order, but as JS objects are unordered, I can't access that ordering when processing the JSON output using JS.

povik, 2023-10-02:
Looking at the source of the JSON backend, it doesn't seem to imprint that information in any way. I see two options: adding an explicit `port_id` attribute or similar on the port object in the JSON output; finding or adding a command that will transfer the port number onto an attribute of the wire, which is then part of the JSON output.

By the way, is there anything public that came out of your work on visualizing the netlists? I am asking just in case it would be relevant to an effort to build an interactive viewer of RTLIL (Yosys netlists), which there's some interest in having.

tilk, 2023-10-02:
> By the way, is there anything public that came out of your work on visualizing the netlists? I am asking just in case it would be relevant to an effort to build an interactive viewer of RTLIL (Yosys netlists), which there's some interest in having.

Yes, the result is the [DigitalJS project](https://github.com/tilk/digitaljs/). It's focused on synthesis and it's probably not much use as general RTLIL viewer, however some ideas might be of use.

povik, 2023-10-02:
To double-check: would the feature of retaining the port ordering still be useful these couple of years later?

tilk, 2023-10-02:
Yes, it is useful, but I believe this works okay now (at least in the pretty old version of Yosys I'm currently using; I'm unable to update because of another issue https://github.com/YosysHQ/yosys/issues/3717). The port ordering in the output is OK, and Javascript currently allows to access this ordering, so I can preserve it in the visualization.

povik, 2023-10-02:
Ah, so you can access the serialization-level ordering after all. From looking at the backend code that should work with current Yosys as well.

Would you be OK with closing the issue, or do you want to keep this around as a feature request for having the port order information explicitly imprinted? Going forward I think it's reasonable to expect that the ports will be printed in-order with the JSON backend.

tilk, 2023-10-02:
I believe the problem kind of solved itself thanks to changes in the Javascript language. I'm fine with closing the issue, thank you for your time.

povik, 2023-10-02:
Happy to! Thanks for the link to DigitalJS.

closed 2023-10-02 by povik