YosysHQ/yosys #604: Yosys fails to ignore the following `specify` block

mithro, 2018-08-14:
## Steps to reproduce the issue

Yosys seems to be unable to parse the following input;

```verilog
//K-input Look-Up Table
module LUT_K #(
    //The Look-up Table size (number of inputs)
    parameter K=2, 

    //The lut mask.  
    //Left-most (MSB) bit corresponds to all inputs logic one. 
    //Defaults to always false.
    parameter LUT_MASK={2**K{1'b0}} 
) (
    input [K-1:0] in,
    output out
);

    specify
        (in *> out) = "";
    endspecify

    assign out = LUT_MASK[in];

endmodule
```

It produces the following error;
```
1. Executing Verilog-2005 frontend.
Parsing Verilog input from `./vtr_flow/primitives.v' to AST representation.
./vtr_flow/primitives.v:140: ERROR: syntax error, unexpected '*', expecting '+' or '-' or '='
```

The line it is complaining about is the `(in *> out) = "";` line inside the specify section.

It is my understanding it should be ignoring the whole `specify`/`endspecify` section?

udif, 2018-08-14:
The specify/endspecify BNF is incomplete.
I added some of the constructs, but there are so many constructs, I haven't finished it.

I would have worked further on this but I already have 2 open pull requests touching verilog_parser.y  that are already 2 month old, yet @cliffordwolf for some reason is not pulling or rejecting those.
I will not touch this file further before these are accepted or rejected, as I don't want to handle merge conflicts caused by this.

ZipCPU, 2018-08-15:
@udif,
Can you point me  to those pull requests?
Dan

cliffordwolf, 2018-08-15:
> yet @cliffordwolf for some reason is not pulling or rejecting those.

I just commented on one of them. Thanks for the reminder.. We do not have a comprehensive test suite yet for the whole spectrum of stuff that should be accepted. So reviewing PRs that artificially limit the range of accepted syntax is a time consuming task. (I'd be less reluctant to just merge it if we wouldn't be in a "we maybe have a release some time the next two months" situation. Otherwise we could just count on the fact that someone will find some code we broke sooner or later, if the PR breaks anything.)

cliffordwolf, 2018-08-15:
@udif both PRs are now merged.

mithro, 2018-08-15:
Just FYI - the definition comes from this file -> https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/vtr_flow/primitives.v

udif, 2018-08-16:
Looking at the file history shows this notation was used back on the 1st commit by @nasrmiad

https://github.com/verilog-to-routing/vtr-verilog-to-routing/commit/b1a3b32e18648554be9013806b33959f7eaa2f81#diff-e637e27f61e0f6d2121f658f69da6e59

However, even this commit is simply a copy from the VTR SVN on googlecode which is dead, and @nasrmiad doesn't seems to be active for the last few years.

Tracing this further to VTR's synthesis tool (ODIN-II) source code:
https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/ODIN_II/SRC/verilog_bison.y
As can be seen on lines 401-403:
```
specify_statement:
    '(' parallel_connection ')' '=' primary ';'							{$$ = newParallelConnection($2, $5, yylineno);}
	;
```
The 'primary' definition boils down to constant numbers and symbols, so I don't see how "" would get parsed correctly.
I also took a look at verilog_lexer.l from the same place, and found no rule that allows \" as a token.

So I really don't know how this could work in the first place.
Have you ever tried using this file with VTR?
To me, this file looks more like a simulation model of an FPGA LUT, a support file to be used by simulators for gate level simulation, and not something that needs to be parsed by synthesis tools. Perhaps no one ever really used it?

mithro, 2018-08-16:
This file is only used in post place and route analysis (when running in simulation).

ZipCPU, 2018-08-17:
I went digging into the System Verilog specification this afternoon.  A string_literal is a valid constant_primary expression.  The referenced code should be interpreted as having a delay of zero.  Yes, it's a strange representation, but as far as I can tell it is legal.

udif, 2018-08-17:
You are right, the simplistic ODIN-II parser does not allow this, but looking even at the older Verilog 2001 EBNF shows this:
path_delay_value -> path_delay_expression -> constant_mintypmax_expression -> constant_expression -> constant_primary -> primary_literal -> string_literal
If I understand the logic correctly, then an empty string literal is a zero width integer that is then padded to the necessary width, hence its value is 0.

I'll try extending the parser further, not because this file was really parsed by VTR, but because it is legal.

mithro, 2018-08-17:
@udif / @ZipCPU - Just FYI this file was used with ModelSim as part of the *output* from VTR to do post-place and route simulation of a design. There is an example of this here -> https://vtr-verilog-to-routing.readthedocs.io/en/latest/tutorials/timing_simulation/index.html#performing-timing-simulation-in-modelsim

udif, 2018-08-20:
I fixed parsing https://github.com/verilog-to-routing/vtr-verilog-to-routing/blob/master/vtr_flow/primitives.v
You do need to use `-defer`:
```
./yosys -p 'read_verilog -defer $VTR/vtr_flow/primitives.v'
```
Because the K parameter in LUT_K has no default value.
Please take look at https://github.com/udif/yosys/tree/udif_specify_round2

mithro, 2018-08-23:
@udif Awesome work!

BTW What is your contact details? I would love to send you an email to discuss some stuff - mine is me@mith.ro

closed 2019-03-14 by cliffordwolf