YosysHQ/yosys #607: 'x' values not uniformly exported to JSON in init values for memories

tilk, 2018-08-16:
The following SystemVerilog code:
```verilog
module rom(input [N-1:0] addr, output [3:0] data);

parameter N = 4;

integer i;

logic [3:0] mem[(1<<N)-1:0];

initial begin
    for (i = 0; i < (1<<N); i = i+1) mem[i] = 4'b1x1x;
end

assign data = mem[addr];

endmodule
```
Gives me the following INIT parameter as exported to JSON:
```json
            "INIT": "1x1x1x1x1x1x1x1x1x1x1x1x1x1x1x1x1x1x1x1x1x1x1x1x1x1x1x1x1x1x1x1x",
```
But if I change the parameter N to 2, I get:
```json
            "INIT": 43690,
```
And therefore, `x` values became `0`s.

I'm working on using the JSON output for simulation, where I use three-valued logic and I would like to preserve `x` values where possible.

daveshah1, 2019-08-07:
With #1241 (which will be merged in the coming days), this now works correctly:

```json
          "parameters": {
            "ABITS": 2,
            "INIT": "1x1x1x1x1x1x1x1x",
            "MEMID": "\\mem",
            "OFFSET": 0,
            "RD_CLK_ENABLE": "0",
            "RD_CLK_POLARITY": "0",
            "RD_PORTS": 1,
            "RD_TRANSPARENT": "0",
            "SIZE": 4,
            "WIDTH": 4,
            "WR_CLK_ENABLE": "0",
            "WR_CLK_POLARITY": "0",
            "WR_PORTS": 0
          },
```

cliffordwolf, 2019-08-11:
We merged #1241 a few days ago. Can this be closed now?

tilk, 2019-08-11:
Yes, this can be closed.

closed 2019-08-11 by tilk