YosysHQ/yosys #609: Need a serials of stable version for packaging.

sequencer, 2018-08-19:
Hello, I wanna package yosys for Arch Linux in community db. And I noticed that since yosys 0.7, stable version disappeared ;(. So if it is possible for yosys to bring back stable versions?

sequencer, 2018-08-20:
And by the way, if I'd like to create a pr to dynamically link `libabc.so`, would it be ok?

cliffordwolf, 2018-08-20:
> So if it is possible for yosys to bring back stable versions?

We are working on it.

> And by the way, if I'd like to create a pr to dynamically link libabc.so, would it be ok?

`LINK_ABC` is for embedded targets. On regular builds Yosys should not link against abc at all, neither statically nor dynamically, but instead just call an external ABC binary. The `ABCEXTERNAL` build variable can be used to set the name of this binary (on debian it is `berkeley-abc` afair).

sequencer, 2018-08-20:
Thank you!
Looking forward to next stable release! ;)

sequencer, 2018-09-09:
ping,
any progress yet?

daveshah1, 2018-12-19:
Yosys 0.8 has been released now.

closed 2018-12-19 by daveshah1