YosysHQ/yosys #612: Segmentation fault on compiling ROM

hutch31, 2018-08-21:
## Steps to reproduce the issue

[zrom.zip](https://github.com/YosysHQ/yosys/files/2308367/zrom.zip)

unzip zrom.zip
yosys
> read_verilog zrom.v
> synth

## Expected behavior

Design should compile OK

## Actual behavior

Segmentation fault during Opt step:

2.6. Executing OPT pass (performing simple optimizations).

2.6.1. Executing OPT_EXPR pass (perform const folding).

2.6.2. Executing OPT_MERGE pass (detect identical cells).
Finding identical cells in module `\zrom'.
Removed a total of 0 cells.

2.6.3. Executing OPT_MUXTREE pass (detect dead branches in mux trees).
Running muxtree optimizer on module \zrom..
  Creating internal representation of mux trees.
  Evaluating internal representation of mux trees.
    Root of a mux tree: $ternary$zrom.v:65475$65472 (pure)

msgctl, 2018-10-19:
I couldn't reproduce this behavior using current *master* branch (commit 76baae4b9), but yosys peaked at ~650M memory usage during OPT_MUXTREE pass, so maybe you just ran out of memory?

hutch31, 2018-10-21:
The failing version was installed from homebrew on mac, and reports as 0.7_1 from brew or 0.7 from inside tool.  I ran same code on linux (reported version as 0.7+605), and it's able to run the synth script, so whatever it was appears to have been fixed in later versions or other architectures.

When I re-ran on my mac under lldb, the call stack goes through eval_mux/eval_mux_port and is about 47k calls deep, so I suspect running out of stack rather than memory (15G of RAM on machine), but this code is not interesting other than potentially exposing bugs.

OK to close.

closed 2019-03-14 by cliffordwolf