YosysHQ/yosys #615: Block RAM inference with negedge clock generates code that can't place

wd5gnr, 2018-08-24:
always @(posedge clka)
  begin
     if (wen) mem[addra]]<=dina; 
   end
always @(negedge clka)
   begin
      doutb<=mem[addrb];
   end

This generates a BLIF, but archane-PNR says:
demo.blif:2026: fatal error: unknown formal `RCLKN'

Looking in the BLIF this is -- no surprise -- in the SB_RAM40_4KNR macro.

daveshah1, 2018-08-24:
This works fine for me with latest Yosys and arachne-pnr, are you sure you are using the latest git versions of each (and not whatever archaeological version distribution packages tend to be)?

If this still doesn't work, please provide a full synthesisable example and the exact commands you are running so it can be investigated further.

wd5gnr, 2018-08-24:
Wow, that was fast. This is with the latest Yosys, but I have not updated Arachne is a while. I will have a look. Thanks.

wd5gnr, 2018-08-24:
Yes, sorry. Updating Arachne did the trick. My bad.

closed 2018-08-24 by wd5gnr

daveshah1, 2018-08-24:
No problem! Glad it is working now