YosysHQ/yosys #622: Duplication of Drivers in a blif generated by Yosys

qaarah, 2018-09-06:
Hi, 
I need a blif file to generated a Miter function. I am generating that blif from yosys that is in the format below.
model top
.inputs io_10_17_0 
.outputs io_0_5_0 io_0_11_1 io_4_0_1 io_5_0_1 io_5_17_1 io_5_17_0 io_6_0_0 io_7_0_0 io_8_0_0 io_8_17_0 io_8_17_1 io_9_0_1 io_13_1_1 io_13_6_0 io_13_7_1 io_13_11_1 io_13_15_1 
 .names [0] io_13_6_0 
1 1
 .names [3] io_13_15_1 
1 1
 .names io_10_17_0 [1] 
1 1
 .names io_10_17_0 [1] 
1 1
 .names io_10_17_0 [1] 
1 1
 .names io_10_17_0 [1] 
1 1
 .names [1771] [290] 
1 1
 .names [1773] [291] 
1 1
 .names [1775] [292] 
1 1
 .names [1777] [293] 

When I run script for Miter Function if gives errors for duplication of drivers. 
Whereas output should not be duplicated. It would be great if someone know anything to solve this issue. 
Yosys Version 0.7
Ubuntu
Thanks 
@mithro

qaarah, 2018-09-14:
@mithro Could you please give some idea. Thanks

cliffordwolf, 2018-09-14:
The new issue template asked you to provide "Steps to reproduce the issue". You didn't. I'm, not sure what your expectations is, but obviously I can't  do anything if I can't reproduce the issue, no I'm closing this now.

closed 2018-09-14 by cliffordwolf