YosysHQ/yosys #624: Cannot find gate "SB_LUT4" in the library. Reading network from file has failed.

qaarah, 2018-09-10:
Hi,

I am trying to compute miter using yosys-abc on a yosys generated blif (synth_ice40 used). But I get the following error.

Creating the full equivalency checking miter in CNF...UC Berkeley, ABC 1.01 (compiled Jun 29 2018 17:10:38)

abc 01> miter /home/qazi/Desktop/Qazi/SFBPrototype/demonstrator/build/example_spe

ecification.blif /home/qazi/Desktop/Qazi/SFBPrototype/demonstrator/build/example_

_techmapped.blif; drw -v; write_cnf /home/qazi/Desktop/Qazi/SFBPrototype/demonstr

rator/build/example_fulleq.cnf; print_stats; quit

Line 11: Skipping line ".attr src "/home/qazi/Desktop/Qazi/SFBPrototype/demonstrator/producerfiles/normal/example.v:24"".

Line 12: Skipping line ".param LUT_INIT 1111000111100010".

Line 14: Skipping line ".attr src "/home/qazi/Desktop/Qazi/SFBPrototype/demonstrator/producerfiles/normal/example.v:23"".

Line 15: Skipping line ".param LUT_INIT 1110101011010100".

Line 17: Skipping line ".attr src "/home/qazi/Desktop/Qazi/SFBPrototype/demonstrator/producerfiles/normal/example.v:25"".

Line 18: Skipping line ".param LUT_INIT 1111000111010100".

Line 20: Skipping line ".attr src "/home/qazi/Desktop/Qazi/SFBPrototype/demonstrator/producerfiles/normal/example.v:22"".

Line 21: Skipping line ".param LUT_INIT 1110101011100010".

Line 23: Skipping line ".attr src "/home/qazi/Desktop/Qazi/SFBPrototype/demonstrator/producerfiles/normal/example.v:32"".

Line 24: Skipping line ".param LUT_INIT 0110101011010101".

Line 26: Skipping line ".attr src "/home/qazi/Desktop/Qazi/SFBPrototype/demonstrator/producerfiles/normal/example.v:19"".

Line 27: Skipping line ".param LUT_INIT 0110101011010101".

Line 29: Skipping line ".attr src "/home/qazi/Desktop/Qazi/SFBPrototype/demonstrator/producerfiles/normal/example.v:18"".

Line 30: Skipping line ".param LUT_INIT 0111000111100011".

Line 32: Skipping line ".attr src "/home/qazi/Desktop/Qazi/SFBPrototype/demonstrator/producerfiles/normal/example.v:20"".

Line 33: Skipping line ".param LUT_INIT 0110101011100011".

Line 35: Skipping line ".attr src "/home/qazi/Desktop/Qazi/SFBPrototype/demonstrator/producerfiles/normal/example.v:17"".

Line 36: Skipping line ".param LUT_INIT 0111000111010101".

Line 38: Skipping line ".attr src "/home/qazi/Desktop/Qazi/SFBPrototype/demonstrator/producerfiles/normal/example.v:29"".

Line 39: Skipping line ".param LUT_INIT 1111000111010101".

Line 41: Skipping line ".attr src "/home/qazi/Desktop/Qazi/SFBPrototype/demonstrator/producerfiles/normal/example.v:28"".

Line 42: Skipping line ".param LUT_INIT 1110101011100011".

Line 44: Skipping line ".attr src "/home/qazi/Desktop/Qazi/SFBPrototype/demonstrator/producerfiles/normal/example.v:30"".

Line 45: Skipping line ".param LUT_INIT 1111000111100011".

Line 47: Skipping line ".attr src "/home/qazi/Desktop/Qazi/SFBPrototype/demonstrator/producerfiles/normal/example.v:27"".

Line 48: Skipping line ".param LUT_INIT 1110101011010101".

Line 50: Skipping line ".attr src "/home/qazi/Desktop/Qazi/SFBPrototype/demonstrator/producerfiles/normal/example.v:14"".

Line 51: Skipping line ".param LUT_INIT 0110101011100010".

Line 53: Skipping line ".attr src "/home/qazi/Desktop/Qazi/SFBPrototype/demonstrator/producerfiles/normal/example.v:13"".

Line 54: Skipping line ".param LUT_INIT 0111000111010100".

Line 56: Skipping line ".attr src "/home/qazi/Desktop/Qazi/SFBPrototype/demonstrator/producerfiles/normal/example.v:15"".

Line 57: Skipping line ".param LUT_INIT 0110101011010100".

Line 59: Skipping line ".attr src "/home/qazi/Desktop/Qazi/SFBPrototype/demonstrator/producerfiles/normal/example.v:12"".

Line 60: Skipping line ".param LUT_INIT 0111000111100010".

**Line 10: **The current library is not available.****

Reading network from file has failed.

abc 01> ***EOF***

I think the Format of that blif is not supported. Can I have some other format in which can be read by abc? The current for mat of my blif is as follows.

.model example

.inputs a

.outputs zro_zro zro_not zro_buf zro_one not_zro not_not not_buf not_one buf_zro buf_not buf_buf buf_one one_zro one_not one_buf one_one magic_inverter

.names $false

.names $true

1

.names $undef

.gate SB_LUT4 I0=a I1=a I2=a I3=a O=buf_buf

.attr src "/home/qazi/Desktop/Qazi/SFBPrototype/demonstrator/producerfiles/normal/example.v:24"

.param LUT_INIT 1111000111100010

.gate SB_LUT4 I0=a I1=a I2=a I3=a O=buf_not

.attr src "/home/qazi/Desktop/Qazi/SFBPrototype/demonstrator/producerfiles/normal/example.v:23"

.param LUT_INIT 1110101011010100

.gate SB_LUT4 I0=a I1=a I2=a I3=a O=buf_one

.attr src "/home/qazi/Desktop/Qazi/SFBPrototype/demonstrator/producerfiles/normal/example.v:25"

.param LUT_INIT 1111000111010100

.gate SB_LUT4 I0=a I1=a I2=a I3=a O=buf_zro

.attr src "/home/qazi/Desktop/Qazi/SFBPrototype/demonstrator/producerfiles/normal/example.v:22"

.param LUT_INIT 1110101011100010

.gate SB_LUT4 I0=a I1=a I2=a I3=a O=magic_inverter

.attr src "/home/qazi/Desktop/Qazi/SFBPrototype/demonstrator/producerfiles/normal/example.v:32"

.param LUT_INIT 0110101011010101

.gate SB_LUT4 I0=a I1=a I2=a I3=a O=not_buf

.attr src "/home/qazi/Desktop/Qazi/SFBPrototype/demonstrator/producerfiles/normal/example.v:19"

.param LUT_INIT 0110101011010101

.gate SB_LUT4 I0=a I1=a I2=a I3=a O=not_not

.attr src "/home/qazi/Desktop/Qazi/SFBPrototype/demonstrator/producerfiles/normal/example.v:18"

.param LUT_INIT 0111000111100011

.gate SB_LUT4 I0=a I1=a I2=a I3=a O=not_one

.attr src "/home/qazi/Desktop/Qazi/SFBPrototype/demonstrator/producerfiles/normal/example.v:20"

.param LUT_INIT 0110101011100011

.gate SB_LUT4 I0=a I1=a I2=a I3=a O=not_zro

.attr src "/home/qazi/Desktop/Qazi/SFBPrototype/demonstrator/producerfiles/normal/example.v:17"

.param LUT_INIT 0111000111010101

.gate SB_LUT4 I0=a I1=a I2=a I3=a O=one_buf

.attr src "/home/qazi/Desktop/Qazi/SFBPrototype/demonstrator/producerfiles/normal/example.v:29"

.param LUT_INIT 1111000111010101

.gate SB_LUT4 I0=a I1=a I2=a I3=a O=one_not

.attr src "/home/qazi/Desktop/Qazi/SFBPrototype/demonstrator/producerfiles/normal/example.v:28"

.param LUT_INIT 1110101011100011

.gate SB_LUT4 I0=a I1=a I2=a I3=a O=one_one

.attr src "/home/qazi/Desktop/Qazi/SFBPrototype/demonstrator/producerfiles/normal/example.v:30"

.param LUT_INIT 1111000111100011

.gate SB_LUT4 I0=a I1=a I2=a I3=a O=one_zro

.attr src "/home/qazi/Desktop/Qazi/SFBPrototype/demonstrator/producerfiles/normal/example.v:27"

.param LUT_INIT 1110101011010101

.gate SB_LUT4 I0=a I1=a I2=a I3=a O=zro_buf

.attr src "/home/qazi/Desktop/Qazi/SFBPrototype/demonstrator/producerfiles/normal/example.v:14"

.param LUT_INIT 0110101011100010

.gate SB_LUT4 I0=a I1=a I2=a I3=a O=zro_not

.attr src "/home/qazi/Desktop/Qazi/SFBPrototype/demonstrator/producerfiles/normal/example.v:13"

.param LUT_INIT 0111000111010100

.gate SB_LUT4 I0=a I1=a I2=a I3=a O=zro_one

.attr src "/home/qazi/Desktop/Qazi/SFBPrototype/demonstrator/producerfiles/normal/example.v:15"

.param LUT_INIT 0110101011010100.

Any help will be appreciated.

Thanks

cliffordwolf, 2018-09-10:
Looks like you are writing a BLIF file with `write_blif -attr -param`. The options `-attr` and `-param` enable BLIF extensions that aren't supported by ABC.

closed 2018-09-10 by cliffordwolf

qaarah, 2018-09-10:
@cliffordwolf Thanks, Yes I am using synth_ice40 which includes -attr and -param. But in my flow, If I use just "-gates" option, VPR starts unlimited placement at one stage that causes error. Anyway, ABC can compute the file without -attr and -param? 
Thanks

qaarah, 2018-09-11:
Hi @cliffordwolf, 
I know you have closed the issue. But I have same problem, I mean writing blif without -attr and -param, but ABC still has the same error. Cannot find gate "SB_LUT4" in the library.
What to do for that?
Thanks