YosysHQ/yosys #630: 3d27c1cc breaks wire assignment in a function

jhol, 2018-09-16:
## Steps to reproduce the issue

```verilog
module top(input in, output out);

function func;
  input arg;
  func = arg;
endfunction

assign out = func(in);

endmodule
```

```
yosys -q -p "synth_ice40 -blif demo.blif" demo.v
```

## Expected behavior

In Yosys 0.7, and commits before `3d27c1cc`, yosys completes the silently.

## Actual behavior

After `3d27c1cc` yosys outputs the following warning:

```
Warning: wire '$func$\func$demo.v:8$2$\arg' is assigned in a block at demo.v:4.
```

udif, 2018-09-16:
Hm... that's my commit :-(

The reason I made this a warning only is exactly for this reason - not to break designs by unexpected bugs.
It also shows the existing tests are lacking.
I will try taking a look at this tomorrow.

jhol, 2018-09-16:
No worries! Thanks for all that you're doing.

jhol, 2018-09-16:
Fortunately the warning goes away is you revert the merge with...

```
git revert -m 1 3d27c1cc
```

jhol, 2018-09-18:
Makes sense. I'm not advocating for these patches to be reverted - just that I have reverted them locally in order to proceed with the project I'm working on.

udif, 2018-09-24:
I pushed a new PR, #638 , to fix this.

closed 2018-10-17 by cliffordwolf