YosysHQ/yosys #635: Error Parsing SystemVerilog

lvcargnini, 2018-09-19:
## Steps to reproduce the issue

Below it is the transcript from my command line, just try to synth the dff and the register based in the dff, described using SV .

## Expected behavior

*Please describe the behavior you would have expected from the tool.*
the following should be observed:
 yosys

 /----------------------------------------------------------------------------\
 |                                                                            |
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |                                                                            |
 |  Copyright (C) 2012 - 2018  Clifford Wolf <clifford@clifford.at>           |
 |                                                                            |
 |  Permission to use, copy, modify, and/or distribute this software for any  |
 |  purpose with or without fee is hereby granted, provided that the above    |
 |  copyright notice and this permission notice appear in all copies.         |
 |                                                                            |
 |  THE SOFTWARE IS PROVIDED "AS IS" AND THE AUTHOR DISCLAIMS ALL WARRANTIES  |
 |  WITH REGARD TO THIS SOFTWARE INCLUDING ALL IMPLIED WARRANTIES OF          |
 |  MERCHANTABILITY AND FITNESS. IN NO EVENT SHALL THE AUTHOR BE LIABLE FOR   |
 |  ANY SPECIAL, DIRECT, INDIRECT, OR CONSEQUENTIAL DAMAGES OR ANY DAMAGES    |
 |  WHATSOEVER RESULTING FROM LOSS OF USE, DATA OR PROFITS, WHETHER IN AN     |
 |  ACTION OF CONTRACT, NEGLIGENCE OR OTHER TORTIOUS ACTION, ARISING OUT OF   |
 |  OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.            |
 |                                                                            |
 \----------------------------------------------------------------------------/

 Yosys 0.7+663 (git sha1 799c4248, clang 6.0.1-svn334776-1~exp1~20180830180435.102 -fPIC -Os)

yosys> read_verilog -sv ../src/sv/dff.sv 
1. Executing Verilog-2005 frontend.
Parsing SystemVerilog input from `../src/sv/dff.sv' to AST representation.
Generating RTLIL representation for module `\dff'.
Successfully finished Verilog frontend.

yosys> read_verilog -sv ../src/sv/register.sv 

2. Executing Verilog-2005 frontend.
Parsing SystemVerilog input from `../src/sv/register.sv' to AST representation.
../src/sv/register.sv:4: ERROR: syntax error
[files.zip](https://github.com/YosysHQ/yosys/files/2398719/files.zip)

## Actual behavior

It should merely synthesize the file. Despite I don't expect the tool to support the entirety of SystemVerilog fully, however, I do expect it to be able to parse any SV file.

AlexDaniel, 2018-09-19:
FWIW the workaround is probably this:
```verilog
parameter XLEN = 4;
```

lvcargnini, 2018-09-19:
ok, but that would break the code everywhere else, wouldn't?
And what to do when we have like two or more parameters, that are separated by ',' ??
```
parameter XLEN=4, 
parameter YADAYADA = 12 
```

cliffordwolf, 2018-09-23:
Omitting the `parameter` keyword in verilog-2001 style param decls in SV mode is now supported in eb452ff.

> Despite I don't expect the tool to support the entirety of SystemVerilog fully, however, I do expect it to be able to parse any SV file.

Even parsing the entirety of SV is not the aim for `read_verilog -sv`.

closed 2018-09-23 by cliffordwolf