YosysHQ/yosys #636: Memory exhausted error on verilog parsing

rafaeltp, 2018-09-20:
## Steps to reproduce the issue

I am running read_verilog on a netlist file (automatically generated by a commercial tool). 
A minimalist code to reproduce is:

```verilog
module foo();

  wire a = {n15312, n15311, n15310, n15309, n15308, n15307, n15306, n15305, n15304, n15303, n15302, n15301, n15300, n15299, n15298, n15297, n15296, n15295, n15294, n15293, n15292, n15291, n15290, n15289, n15288, n8400, n15286, n15285, n15284, n15283, n15282, n15281, n15280, n15279, n15278, n15277, n15276, n15275, n15274, n15273, n15272, n15271, n15270, n15269, n15268, n15267, n15266, n15265, n15264, n15263, n15262, n8399, n15260, n15259, n15258, n15257, n15256, n15255, n15254, n15253, n15252, n15251, n15250, n15249, n15248, n15247, n15246, n15245, n15244, n15243, n15242, n15241, n15240, n15239, n15238, n15237, n15236, n8398, n15234, n15233, n15232, n15231, n15230, n15229, n15228, n15227, n15226, n15225, n15224, n15223, n15222, n15221, n15220, n15219, n15218, n15217, n15216, n15215, n15214, n15213, n15212, n15211, n15210, n8397, n15208, n15207, n15206, n15205, n15204, n15203, n15202, n15201, n15200, n15199, n15198, n15197, n15196, n15195, n15194, n15193, n15192, n15191, n15190, n15189, n15188, n15187, n15186, n15185, n15184, n8396, n15182, n15181, n15180, n15179, n15178, n15177, n15176, n15175, n15174, n15173, n15172, n15171, n15170, n15169, n15168, n15167, n15166, n15165, n15164, n15163, n15162, n15161, n15160, n15159, n15158, n8395, n15156, n15155, n15154, n15153, n15152, n15151, n15150, n15149, n15148, n15147, n15146, n15145, n15144, n15143, n15142, n15141, n15140, n15139, n15138, n15137, n15136, n15135, n15134, n15133, n15132, n8394, n15130, n15129, n15128, n15127, n15126, n15125, n15124, n15123, n15122, n15121, n15120, n15119, n15118, n15117, n15116, n15115, n15114, n15113, n15112, n15111, n15110, n15109, n15108, n15107, n15106, n8393, n15104, n15103, n15102, n15101, n15100, n15099, n15098, n15097, n15096, n15095, n15094, n15093, n15092, n15091, n15090, n15089, n15088, n15087, n15086, n15085, n15084, n15083, n15082, n15081, n15080, n8392, n15078, n15077, n15076, n15075, n15074, n15073, n15072, n15071, n15070, n15069, n15068, n15067, n15066, n15065, n15064, n15063, n15062, n15061, n15060, n15059, n15058, n15057, n15056, n15055, n15054, n8391, n15052, n15051, n15050, n15049, n15048, n15047, n15046, n15045, n15044, n15043, n15042, n15041, n15040, n15039, n15038, n15037, n15036, n15035, n15034, n15033, n15032, n15031, n15030, n15029, n15028, n8390, n15026, n15025, n15024, n15023, n15022, n15021, n15020, n15019, n15018, n15017, n15016, n15015, n15014, n15013, n15012, n15011, n15010, n15009, n15008, n15007, n15006, n15005, n15004, n15003, n15002, n8389, n15000, n14999, n14998, n14997, n14996, n14995, n14994, n14993, n14992, n14991, n14990, n14989, n14988, n14987, n14986, n14985, n14984, n14983, n14982, n14981, n14980, n14979, n14978, n14977, n14976, n8388, n14974, n14973, n14972, n14971, n14970, n14969, n14968, n14967, n14966, n14965, n14964, n14963, n14962, n14961, n14960, n14959, n14958, n14957, n14956, n14955, n14954, n14953, n14952, n14951, n14950, n8387, n14948, n14947, n14946, n14945, n14944, n14943, n14942, n14941, n14940, n14939, n14938, n14937, n14936, n14935, n14934, n14933, n14932, n14931, n14930, n14929, n14928, n14927, n14926, n14925, n14924, n8386, n14922, n14921, n14920, n14919, n14918, n14917, n14916, n14915, n14914, n14913, n14912, n14911, n14910, n14909, n14908, n14907, n14906, n14905, n14904, n14903, n14902, n14901, n14900, n14899, n14898, n8385, n14896, n14895, n14894, n14893, n14892, n14891, n14890, n14889, n14888, n14887, n14886, n14885, n14884, n14883, n14882, n14881, n14880, n14879, n14878, n14877, n14876, n14875, n14874, n14873, n14872, n8384, n14870, n14869, n14868, n14867, n14866, n14865, n14864, n14863, n14862, n14861, n14860, n14859, n14858, n14857, n14856, n14855, n14854, n14853, n14852, n14851, n14850, n14849, n14848, n14847, n14846, n8383, n14844, n14843, n14842, n14841, n14840, n14839, n14838, n14837, n14836, n14835, n14834, n14833, n14832, n14831, n14830, n14829, n14828, n14827, n14826, n14825, n14824, n14823, n14822, n14821, n14820, n8382, n14818, n14817, n14816, n14815, n14814, n14813, n14812, n14811, n14810, n14809, n14808, n14807, n14806, n14805, n14804, n14803, n14802, n14801, n14800, n14799, n14798, n14797, n14796, n14795, n14794, n8381, n14792, n14791, n14790, n14789, n14788, n14787, n14786, n14785, n14784, n14783, n14782, n14781, n14780, n14779, n14778, n14777, n14776, n14775, n14774, n14773, n14772, n14771, n14770, n14769, n14768, n8380, n14766, n14765, n14764, n14763, n14762, n14761, n14760, n14759, n14758, n14757, n14756, n14755, n14754, n14753, n14752, n14751, n14750, n14749, n14748, n14747, n14746, n14745, n14744, n14743, n14742, n8379, n14740, n14739, n14738, n14737, n14736, n14735, n14734, n14733, n14732, n14731, n14730, n14729, n14728, n14727, n14726, n14725, n14724, n14723, n14722, n14721, n14720, n14719, n14718, n14717, n14716, n8378, n14714, n14713, n14712, n14711, n14710, n14709, n14708, n14707, n14706, n14705, n14704, n14703, n14702, n14701, n14700, n14699, n14698, n14697, n14696, n14695, n14694, n14693, n14692, n14691, n14690, n8377, n14688, n14687, n14686, n14685, n14684, n14683, n14682, n14681, n14680, n14679, n14678, n14677, n14676, n14675, n14674, n14673, n14672, n14671, n14670, n14669, n14668, n14667, n14666, n14665, n14664, n8376, n14662, n14661, n14660, n14659, n14658, n14657, n14656, n14655, n14654, n14653, n14652, n14651, n14650, n14649, n14648, n14647, n14646, n14645, n14644, n14643, n14642, n14641, n14640, n14639, n14638, n8375, n14636, n14635, n14634, n14633, n14632, n14631, n14630, n14629, n14628, n14627, n14626, n14625, n14624, n14623, n14622, n14621, n14620, n14619, n14618, n14617, n14616, n14615, n14614, n14613, n14612, n8374, n14610, n14609, n14608, n14607, n14606, n14605, n14604, n14603, n14602, n14601, n14600, n14599, n14598, n14597, n14596, n14595, n14594, n14593, n14592, n14591, n14590, n14589, n14588, n14587, n14586, n8373, n14584, n14583, n14582, n14581, n14580, n14579, n14578, n14577, n14576, n14575, n14574, n14573, n14572, n14571, n14570, n14569, n14568, n14567, n14566, n14565, n14564, n14563, n14562, n14561, n14560, n8372, n14558, n14557, n14556, n14555, n14554, n14553, n14552, n14551, n14550, n14549, n14548, n14547, n14546, n14545, n14544, n14543, n14542, n14541, n14540, n14539, n14538, n14537, n14536, n14535, n14534, n8371, n14532, n14531, n14530, n14529, n14528, n14527, n14526, n14525, n14524, n14523, n14522, n14521, n14520, n14519, n14518, n14517, n14516, n14515, n14514, n14513, n14512, n14511, n14510, n14509, n14508, n8370, n14506, n14505, n14504, n14503, n14502, n14501, n14500, n14499, n14498, n14497, n14496, n14495, n14494, n14493, n14492, n14491, n14490, n14489, n14488, n14487, n14486, n14485, n14484, n14483, n14482, n8369, n14480, n14479, n14478, n14477, n14476, n14475, n14474, n14473, n14472, n14471, n14470, n14469, n14468, n14467, n14466, n14465, n14464, n14463, n14462, n14461, n14460, n14459, n14458, n14457, n14456, n8368, n14454, n14453, n14452, n14451, n14450, n14449, n14448, n14447, n14446, n14445, n14444, n14443, n14442, n14441, n14440, n14439, n14438, n14437, n14436, n14435, n14434, n14433, n14432, n14431, n14430, n8367, n14428, n14427, n14426, n14425, n14424, n14423, n14422, n14421, n14420, n14419, n14418, n14417, n14416, n14415, n14414, n14413, n14412, n14411, n14410, n14409, n14408, n14407, n14406, n14405, n14404, n8366, n14402, n14401, n14400, n14399, n14398, n14397, n14396, n14395, n14394, n14393, n14392, n14391, n14390, n14389, n14388, n14387, n14386, n14385, n14384, n14383, n14382, n14381, n14380, n14379, n14378, n8365, n14376, n14375, n14374, n14373, n14372, n14371, n14370, n14369, n14368, n14367, n14366, n14365, n14364, n14363, n14362, n14361, n14360, n14359, n14358, n14357, n14356, n14355, n14354, n14353, n14352, n8364, n14350, n14349, n14348, n14347, n14346, n14345, n14344, n14343, n14342, n14341, n14340, n14339, n14338, n14337, n14336, n14335, n14334, n14333, n14332, n14331, n14330, n14329, n14328, n14327, n14326, n8363, n14324, n14323, n14322, n14321, n14320, n14319, n14318, n14317, n14316, n14315, n14314, n14313, n14312, n14311, n14310, n14309, n14308, n14307, n14306, n14305, n14304, n14303, n14302, n14301, n14300, n8362, n14298, n14297, n14296, n14295, n14294, n14293, n14292, n14291, n14290, n14289, n14288, n14287, n14286, n14285, n14284, n14283, n14282, n14281, n14280, n14279, n14278, n14277, n14276, n14275, n14274, n8361, n14272, n14271, n142 [... cut ...] 9, n8668, n8667, n8666, n8665, n8664, n8663, n8662, n8661, n8660, n8659, n8658, n8145};

endmodule
```

(Note: I know the sample doesn't make sense as a verilog module, it was extracted from a much larger file)

## Expected behavior

I just expect to be able to parse the line :-)

## Actual behavior

I am getting the following error message:

```
1. Executing Verilog-2005 frontend.
Parsing Verilog input from `foo.v' to AST representation.
ERROR: Parser error in line foo.v:4: memory exhausted
```

tomverbeure, 2018-10-02:
The error happens because the stack of the verilog parser is exhausted. 

The default depth is 10000. 

Your specific example can be fixed by making the following change in the source code, which increases the stack from 10000 to 15000:

```
diff --git a/frontends/verilog/verilog_parser.y b/frontends/verilog/verilog_parser.y
index 16cac14..d3c59d0 100644
--- a/frontends/verilog/verilog_parser.y
+++ b/frontends/verilog/verilog_parser.y
@@ -39,6 +39,8 @@
 #include "frontends/verilog/verilog_frontend.h"
 #include "kernel/log.h"

+#define YYMAXDEPTH 15000
+
 USING_YOSYS_NAMESPACE
```

But ultimately, if your tool creates lines that are 51000 characters long, it might as well create lines that are 510000 characters long on a larger design, and you'll be back to square one.

daveshah1, 2019-05-18:
No longer seeing the error on latest master, suspect this has been fixed by #1017.

closed 2019-05-18 by daveshah1