YosysHQ/yosys #642: Yosys segfault on $past(X,parameter_value-1)

ZipCPU, 2018-09-26:
## Steps to reproduce the issue

Run yosys, and read the attached file using read_verilog -formal mcve.v.

    module testing(clk, in, out);
        parameter DEPTH=10;
        input wire clk, in;
        output reg out;
    
        always @(posedge clk)
            out <= $past(in,DEPTH-1);
    endmodule

## Expected behavior

I would have expected this to implement $past(in,9) without the segfault, or if this wasn't supported it should provide a description of what caused the segfault and why.

## Actual behavior

Segfault

## Analysis

I traced this bug down as far as line [1788 of simplify.cc](https://github.com/YosysHQ/yosys/blob/master/frontends/ast/simplify.cc#L1788).  That's the line where the second argument of $past is processed.  The value returned is given as the constant zero, rather than the constant nine.  When the code the enters the [loop at 1803](https://github.com/YosysHQ/yosys/blob/master/frontends/ast/simplify.cc#L1803), the number of iterations is zero.  As a result, outreg is left as NULL and the segfault takes place on [line 1834](https://github.com/YosysHQ/yosys/blob/master/frontends/ast/simplify.cc#L1834).

Dan

ZipCPU, 2018-09-26:
I just tried the code using an integer instead of a parameter.  $past(in,9) works, whereas $past(in,10-1) does not.

promach, 2018-09-29:
@ZipCPU Why [line 1788 of simplify.cc](https://github.com/YosysHQ/yosys/blob/master/frontends/ast/simplify.cc#L1788) returns constant zero ?

cliffordwolf, 2018-09-30:
Thanks for reporting this issue. Fixed in 9f9fe94.

closed 2018-09-30 by cliffordwolf