YosysHQ/yosys #643: libreadline.so.6 issue in ubuntu 18.04

7segv, 2018-09-26:
I upgrade my system to 18.04 and when trying to run yosys I get issue below:

> yosys: error while loading shared libraries: libreadline.so.6: cannot open shared object file: No such file or directory

I try to uninstall libreadline-dev with
>  sudo apt-get purge libreadline-dev

and install the libreadline.6 with
> sudo apt-get install libreadline6-dev

but the system install the latest version (libreadline-dev_7.0-3) again:
> Note, selecting 'libreadline-dev' instead of 'libreadline6-dev'

Is there a way to fix the issue?

daveshah1, 2018-09-27:
Did you rebuild Yosys after updating Ubuntu?

cliffordwolf, 2019-03-12:
Closing as issues seems to be abandoned by OP. Simply rebuilding Yosys, as suggested by daveshah1, should fix the problem.

closed 2019-03-12 by cliffordwolf