YosysHQ/yosys #652: Error IDELAYCTRL missing for IODELAYs when using Yosys + Vivado with basic LiteX SoC

mithro, 2018-10-05:
## Steps to reproduce the issue

You will need the latest Yosys with https://github.com/YosysHQ/yosys/pull/650 included and Vivado with WebPack license.

Then clone [this temporary repository](https://github.com/mithro/litex-buildenv-arty-yosys-xilinx), then try a build.

```shell
git clone https://github.com/mithro/litex-buildenv-arty-yosys-xilinx.git
cd litex-buildenv-arty-yosys-xilinx
./build_top.sh
```

I have also committed the output that I have received. I'm using Vivado 2017.3 and Yosys `0.7+663 (git sha1 dc77ed1e)` with the patch in pull request #650. (When I get a chance I'll use latest Vivado and Yosys.)

## Expected behavior

Produces a working bitstream for this design (as using Vivado's inbuilt synthesis for this design does).

You can compare the behaviour to a build which only uses Vivado at https://github.com/timvideos/HDMI2USB-firmware-prebuilt/tree/master/archive/master/v0.0.4-356-ga1093ac/arty/base/lm32/gateware

## Actual behavior

When trying to use Yosys for the synthesis stage as part of a Vivado flow for an Artix 7 that uses IODELAY objects, the build fails with;
```
Time (s): cpu = 00:00:06 ; elapsed = 00:00:03 . Memory (MB): peak = 2493.836 ; gain = 416.562 ; free physical = 701 ; free virtual = 32415
INFO: [Common 17-83] Releasing license: Implementation
21 Infos, 0 Warnings, 0 Critical Warnings and 0 Errors encountered.
opt_design completed successfully
opt_design: Time (s): cpu = 00:00:10 ; elapsed = 00:00:06 . Memory (MB): peak = 2493.836 ; gain = 489.594 ; free physical = 701 ; free virtual = 32415
# place_design
Command: place_design
Attempting to get a license for feature 'Implementation' and/or device 'xc7a35t'
INFO: [Common 17-349] Got license for feature 'Implementation' and/or device 'xc7a35t'
INFO: [DRC 23-27] Running DRC with 8 threads
INFO: [Vivado_Tcl 4-198] DRC finished with 0 Errors
INFO: [Vivado_Tcl 4-199] Please refer to the DRC report (report_drc) for more information.
Running DRC as a precondition to command place_design
INFO: [DRC 23-27] Running DRC with 8 threads
ERROR: [DRC PLIDC-10] IDELAYCTRL missing for IODELAYs: There are 16 IDELAY/ODELAY/IODELAY cells in the design which requires IDelayCtrl, but there is no IDelayCtrl cell
INFO: [Vivado_Tcl 4-198] DRC finished with 1 Errors
INFO: [Vivado_Tcl 4-199] Please refer to the DRC report (report_drc) for more information.
ERROR: [Vivado_Tcl 4-23] Error(s) found during DRC. Placer not run.
INFO: [Common 17-83] Releasing license: Implementation
8 Infos, 0 Warnings, 0 Critical Warnings and 2 Errors encountered.
place_design failed
ERROR: [Common 17-39] 'place_design' failed due to earlier errors.

    while executing
"place_design"
    (file "top.tcl" line 9)
```

The working Vivado flow looks like the following;
```tcl

create_project -force -name top -part xc7a35t-csg324-1
add_files {/home/travis/build/timvideos/HDMI2USB-litex-firmware/third_party/litex/litex/soc/cores/cpu/lm32/verilog/submodule/rtl/lm32_cpu.v}
set_property library work [get_files {/home/travis/build/timvideos/HDMI2USB-litex-firmware/third_party/litex/litex/soc/cores/cpu/lm32/verilog/submodule/rtl/lm32_cpu.v}]
add_files {/home/travis/build/timvideos/HDMI2USB-litex-firmware/third_party/litex/litex/soc/cores/cpu/lm32/verilog/submodule/rtl/lm32_instruction_unit.v}
set_property library work [get_files {/home/travis/build/timvideos/HDMI2USB-litex-firmware/third_party/litex/litex/soc/cores/cpu/lm32/verilog/submodule/rtl/lm32_instruction_unit.v}]
add_files {/home/travis/build/timvideos/HDMI2USB-litex-firmware/third_party/litex/litex/soc/cores/cpu/lm32/verilog/submodule/rtl/lm32_decoder.v}
set_property library work [get_files {/home/travis/build/timvideos/HDMI2USB-litex-firmware/third_party/litex/litex/soc/cores/cpu/lm32/verilog/submodule/rtl/lm32_decoder.v}]
add_files {/home/travis/build/timvideos/HDMI2USB-litex-firmware/third_party/litex/litex/soc/cores/cpu/lm32/verilog/submodule/rtl/lm32_load_store_unit.v}
set_property library work [get_files {/home/travis/build/timvideos/HDMI2USB-litex-firmware/third_party/litex/litex/soc/cores/cpu/lm32/verilog/submodule/rtl/lm32_load_store_unit.v}]
add_files {/home/travis/build/timvideos/HDMI2USB-litex-firmware/third_party/litex/litex/soc/cores/cpu/lm32/verilog/submodule/rtl/lm32_adder.v}
set_property library work [get_files {/home/travis/build/timvideos/HDMI2USB-litex-firmware/third_party/litex/litex/soc/cores/cpu/lm32/verilog/submodule/rtl/lm32_adder.v}]
add_files {/home/travis/build/timvideos/HDMI2USB-litex-firmware/third_party/litex/litex/soc/cores/cpu/lm32/verilog/submodule/rtl/lm32_addsub.v}
set_property library work [get_files {/home/travis/build/timvideos/HDMI2USB-litex-firmware/third_party/litex/litex/soc/cores/cpu/lm32/verilog/submodule/rtl/lm32_addsub.v}]
add_files {/home/travis/build/timvideos/HDMI2USB-litex-firmware/third_party/litex/litex/soc/cores/cpu/lm32/verilog/submodule/rtl/lm32_logic_op.v}
set_property library work [get_files {/home/travis/build/timvideos/HDMI2USB-litex-firmware/third_party/litex/litex/soc/cores/cpu/lm32/verilog/submodule/rtl/lm32_logic_op.v}]
[... 66 lines cut ...]
report_timing_summary -no_header -no_detailed_paths
write_checkpoint -force top_route.dcp
report_route_status -file top_route_status.rpt
report_drc -file top_drc.rpt
report_timing_summary -datasheet -max_paths 10 -file top_timing.rpt
report_power -file top_power.rpt
set_property BITSTREAM.CONFIG.SPI_BUSWIDTH 4 [current_design]
write_bitstream -force top.bit 
write_cfgmem -force -format bin -interface spix4 -size 16 -loadbit "up 0x0 top.bit" -file top.bin
quit
```

Could the issue could be related to failure to apply the following constraints?
```
Parsing XDC File [/tansell/github/timvideos/HDMI2USB-litex-firmware/build/arty_base_lm32/gateware/top.xdc]
WARNING: [Vivado 12-1023] No nets matched for command 'get_nets -filter {mr_ff == TRUE}'. [/tansell/github/timvideos/HDMI2USB-litex-firmware/build/arty_base_lm32/gateware/top.xdc:282]
WARNING: [Vivado 12-180] No cells matched 'get_cells -filter {ars_ff1 == TRUE || ars_ff2 == TRUE}'. [/tansell/github/timvideos/HDMI2USB-litex-firmware/build/arty_base_lm32/gateware/top.xdc:284]
WARNING: [Vivado 12-508] No pins matched 'get_pins -filter {REF_PIN_NAME == PRE} -of [get_cells -filter {ars_ff1 == TRUE || ars_ff2 == TRUE}]'. [/home/tansell/github/timvideos/HDMI2USB-litex-firmware/build/arty_base_lm32/
gateware/top.xdc:284]
WARNING: [Vivado 12-180] No cells matched 'get_cells -filter {ars_ff1 == TRUE}'. [/home/tansell/github/timvideos/HDMI2USB-litex-firmware/build/arty_base_lm32/gateware/top.xdc:286]
WARNING: [Vivado 12-508] No pins matched 'get_pins -filter {REF_PIN_NAME == Q} -of [get_cells -filter {ars_ff1 == TRUE}]'. [/home/tansell/github/timvideos/HDMI2USB-litex-firmware/build/arty_base_lm32/gateware/top.xdc:286]
WARNING: [Vivado 12-180] No cells matched 'get_cells -filter {ars_ff2 == TRUE}'. [/tansell/github/timvideos/HDMI2USB-litex-firmware/build/arty_base_lm32/gateware/top.xdc:286]
WARNING: [Vivado 12-508] No pins matched 'get_pins -filter {REF_PIN_NAME == D} -of [get_cells -filter {ars_ff2 == TRUE}]'. [/home/tansell/github/timvideos/HDMI2USB-litex-firmware/build/arty_base_lm32/gateware/top.xdc:286]
Finished Parsing XDC File [/home/tansell/github/timvideos/HDMI2USB-litex-firmware/build/arty_base_lm32/gateware/top.xdc]
```

mithro, 2018-10-05:
Just confirming that this occurs with the latest Yosys (` Yosys 0.7+677 (git sha1 5f1fea08, gcc 7.3.0-5 -fPIC -Os)`) build from head.

Going to try with the latest Vivado.

mithro, 2018-10-05:
Just confirming that this also fails with latest Vivado (tried with `Xilinx_Vivado_SDK_Web_2018.2_0614_1954_Lin64.bin`).

mithro, 2018-10-05:
I built a version of this SoC without the SPI Flash or DDR and ended up with a `.bit` file! Will test is the file actually works sometime soon.

You can find the output of this minimal SoC at https://github.com/mithro/litex-buildenv-arty-yosys-xilinx/tree/min

mithro, 2018-10-05:
```
# write_checkpoint -force top_route.dcp
INFO: [Timing 38-480] Writing timing data to binary archive.
Writing placer database...
Writing XDEF routing.
Writing XDEF routing logical nets.
Writing XDEF routing special nets.
Write XDEF Complete: Time (s): cpu = 00:00:01 ; elapsed = 00:00:00.37 . Memory (MB): peak = 2471.555 ; gain = 0.000 ; free physical = 5826 ; free virtual = 37222
INFO: [Common 17-1381] The checkpoint '/home/tansell/github/timvideos/HDMI2USB-litex-firmware/build/arty_min_lm32/gateware/top_route.dcp' has been generated.
# report_route_status -file top_route_status.rpt
# report_drc -file top_drc.rpt
Command: report_drc -file top_drc.rpt
INFO: [IP_Flow 19-234] Refreshing IP repositories
INFO: [IP_Flow 19-1704] No user IP repositories specified
INFO: [IP_Flow 19-2313] Loaded Vivado IP repository '/opt/Xilinx/Vivado/2018.2/data/ip'.
INFO: [DRC 23-27] Running DRC with 8 threads
INFO: [Coretcl 2-168] The results of DRC are in file /home/tansell/github/timvideos/HDMI2USB-litex-firmware/build/arty_min_lm32/gateware/top_drc.rpt.
report_drc completed successfully
# report_timing_summary -datasheet -max_paths 10 -file top_timing.rpt
INFO: [Timing 38-91] UpdateTimingParams: Speed grade: -1, Delay Type: min_max.
INFO: [Timing 38-191] Multithreading enabled for timing update using a maximum of 8 CPUs
# report_power -file top_power.rpt
Command: report_power -file top_power.rpt
INFO: [Power 33-23] Power model is not available for DNA_PORT
Running Vector-less Activity Propagation...

Finished Running Vector-less Activity Propagation
1 Infos, 0 Warnings, 0 Critical Warnings and 0 Errors encountered.
report_power completed successfully
# set_property BITSTREAM.CONFIG.SPI_BUSWIDTH 4 [current_design]
# write_bitstream -force top.bit 
Command: write_bitstream -force top.bit
Attempting to get a license for feature 'Implementation' and/or device 'xc7a35t'
INFO: [Common 17-349] Got license for feature 'Implementation' and/or device 'xc7a35t'
Running DRC as a precondition to command write_bitstream
INFO: [DRC 23-27] Running DRC with 8 threads
WARNING: [DRC CFGBVS-1] Missing CFGBVS and CONFIG_VOLTAGE Design Properties: Neither the CFGBVS nor CONFIG_VOLTAGE voltage property is set in the current_design.  Configuration bank voltage select (CFGBVS) must be set to VCCO or GND, and CONFIG_VOLTAGE must be set to the correct configuration voltage, in order to determine the I/O voltage support for the pins in bank 0.  It is suggested to specify these either using the 'Edit Device Properties' function in the GUI or directly in the XDC file using the following syntax:

 set_property CFGBVS value1 [current_design]
 #where value1 is either VCCO or GND

 set_property CONFIG_VOLTAGE value2 [current_design]
 #where value2 is the voltage provided to configuration bank 0

Refer to the device configuration user guide for more information.
INFO: [Vivado 12-3199] DRC finished with 0 Errors, 1 Warnings
INFO: [Vivado 12-3200] Please refer to the DRC report (report_drc) for more information.
INFO: [Designutils 20-2272] Running write_bitstream with 8 threads.
Loading data files...
Loading site data...
Loading route data...
Processing options...
Creating bitmap...
Creating bitstream...
Writing bitstream ./top.bit...
INFO: [Vivado 12-1842] Bitgen Completed Successfully.
INFO: [Project 1-118] WebTalk data collection is enabled (User setting is ON. Install Setting is ON.).
INFO: [Common 17-186] '/home/tansell/github/timvideos/HDMI2USB-litex-firmware/build/arty_min_lm32/gateware/usage_statistics_webtalk.xml' has been successfully sent to Xilinx on Thu Oct  4 18:52:33 2018. For additional details about this file, please refer to the WebTalk help file at /opt/Xilinx/Vivado/2018.2/doc/webtalk_introduction.html.
INFO: [Common 17-83] Releasing license: Implementation
9 Infos, 1 Warnings, 0 Critical Warnings and 0 Errors encountered.
write_bitstream completed successfully
write_bitstream: Time (s): cpu = 00:00:15 ; elapsed = 00:00:27 . Memory (MB): peak = 2502.676 ; gain = 31.121 ; free physical = 5790 ; free virtual = 37183
# write_cfgmem -force -format bin -interface spix4 -size 16 -loadbit "up 0x0 top.bit" -file top.bin
Command: write_cfgmem -force -format bin -interface spix4 -size 16 -loadbit {up 0x0 top.bit} -file top.bin
Creating config memory files...
Creating bitstream load up from address 0x00000000
Loading bitfile top.bit
Writing file ./top.bin
Writing log file ./top.prm
===================================
Configuration Memory information
===================================
File Format        BIN
Interface          SPIX4
Size               16M
Start Address      0x00000000
End Address        0x00FFFFFF

Addr1         Addr2         Date                    File(s)
0x00000000    0x0021728B    Oct  4 18:52:32 2018    top.bit
0 Infos, 0 Warnings, 0 Critical Warnings and 0 Errors encountered.
write_cfgmem completed successfully
# quit
INFO: [Common 17-206] Exiting Vivado at Thu Oct  4 18:52:35 2018...
```

mithro, 2018-10-05:
Might need something like https://www.xilinx.com/support/answers/68753.html

mithro, 2018-10-05:
The issue seems to be that the `top.v` has;
```verilog
IDELAYCTRL IDELAYCTRL(
	.REFCLK(clk200_clk),
	.RST(ic_reset)
);
```

But the output `top.edif` file produced by Yosys does not have any IDELAYCTRL cells?

mithro, 2018-10-05:
```
20.5.3. Executing OPT_CLEAN pass (remove unused cells and wires).
...
  removing unused `\IDELAYCTRL' cell `\IDELAYCTRL'.
  removing unused `\STARTUPE2' cell `\STARTUPE2'.
...
```

mithro, 2018-10-05:
It seems by adding;
```
# hierarchy -top top
# proc; memory; opt; fsm; opt
attrmap -tocase keep -imap keep="true" keep=1 \
    -imap keep="false" keep=0 -remove keep=0
synth_xilinx -top top -edif top.edif
```

And then adding a `(* KEEP *)` to the `IDELAYCTRL` object like so;
```verilog
(* KEEP = "TRUE" *)
IDELAYCTRL IDELAYCTRL(
	.REFCLK(clk200_clk),
	.RST(ic_reset)
);
```

Allows Vivado to generate a bitstream.

I have pushed the latest working code to [the repository here](https://github.com/mithro/litex-buildenv-arty-yosys-xilinx).

mithro, 2018-10-05:
Should the `IDELAYCTRL` primitive always have the keep property?

enjoy-digital, 2018-10-05:
`IDELAYCTRL` is indeed a bit special: it has inputs but no outputs. So what probably happens is that without the `KEEP` attribute Yosys will just remove it and that's the expected behaviour for this kind of modules.

cliffordwolf, 2018-10-05:
> So what probably happens is that without the KEEP attribute Yosys will just remove it and that's the expected behaviour for this kind of modules.

Yep.

However, `IDELAYCTRL` does have an output: `RDY`. It is just unused in this design.

Regarding this script:

```
# hierarchy -top top
# proc; memory; opt; fsm; opt
attrmap -tocase keep -imap keep="true" keep=1 \
    -imap keep="false" keep=0 -remove keep=0
synth_xilinx -top top -edif top.edif
```

You *definitiely* want to run `hierarchy` before running `attrmap`, or `attrmap` will not do its magic in modules that are instantiated with parameters. We had this discussion before: always call `hierarchy` as the first command after reading the sources unles you have a really very good reason not to.

> Should the `IDELAYCTRL` primitive always have the keep property?

Idk. Maybe. But if so then other modules should as well. I'm not going to add a module-level `(* keep *)` to `IDELAYCTRL` only. If you can come up with a reasonable criteria which modules should have `keep` set and provide me with the list of modules from `cells_xtra.v` that matches that criterie then I'll be happy to add the attribute.

(Just to clarify: You would of course not set it automatically on instances, which I think is what your wording implies. You'd just set it once on the module in `cells_xtra.v` and be done with it.)

closed 2018-10-05 by cliffordwolf