YosysHQ/yosys #662: Missing edge-sensitive event for signal

mngr0, 2018-10-14:
Here I have some Verilog code
[error.zip](https://github.com/YosysHQ/yosys/files/2476242/error.zip)
and I am running yosys on it, but it gives an error
`ERROR: Missing edge-sensitive event for this signal!`

that is caused by the very last always block in the code
```
always @(posedge sck, negedge spibus_csn) begin: SPI_SLAVE_LED_SPI_SLAVE_FIFO0_CSN_FALLS
    if (sck) begin
        spi_slave_fifo0_spi_start <= 1'b0;
    end
    else if ((!spibus_csn)) begin
        spi_slave_fifo0_spi_start <= 1'b1;
    end
end
```

I don't understand the problem, if I change to
`always @(posedge sck, posedge spibus_csn) begin: `
it compiles.

This verilog code is generated using MyHDL

cliffordwolf, 2019-03-12:
> I don't understand the problem, if I change to
`always @(posedge sck, posedge spibus_csn) begin:`
it compiles.

Wait, what? `begin:`, including the final `:` but excluding any block name? That's a syntax error in my test, as it should be:

```
Parsing Verilog input from `spi_slave_led.v' to AST representation.
spi_slave_led.v:430: ERROR: syntax error, unexpected TOK_IF, expecting TOK_ID
```

And if I also remove the `:` then I get the same result as with the unmodified code, which is exactly what I would have been expecting. (Yosys doesn't support this kind of flop that's active on many different async reset events.)

forics, 2020-08-24:
> Yosys doesn't support this kind of flop that's active on many different async reset events.

Wander to know why. @cliffordwolf

I got stuck in the same problem. I think blocks with posedge-triggered clock signal and negedge-triggered reset signal are common. Is there a plan to support this case or a workaround?

forics, 2020-08-24:
> Yosys doesn't support this kind of flop that's active on many different async reset events.

Wander to know why. @cliffordwolf

----

Under commit [12132b6](https://github.com/YosysHQ/yosys/commit/12132b6850747aec99715fdfa3184fe3ebefa015), syntax error disappears but proc_dff error still exists. I explore some variants of the code snippet given by @mngr0 .

1. variant A
```
always @(posedge rst, negedge clk) begin
    if (rst) begin
        q <= 1'b0;
    end
    else begin
        q <= 1'b1;  // or some signal din
    end
end
```
got no errors.

2. variant B
```
always @(posedge rst, negedge clk) begin
   q <= rst ? 1'b0 : 1'b1;
end
```
got **ERROR: Missing edge-sensitive event for this signal!**

By the way, I join the discuss because I come across variant B and get the error recently.

mwkmwkmwk, 2020-08-24:
> > Yosys doesn't support this kind of flop that's active on many different async reset events.
> 
> Wander to know why. @cliffordwolf

The code does not correspond to a recognized FF pattern — it has two asynchronous resets, but no clock signal, so it cannot be a sequential element.  It could perhaps be interpreted as a latch, but latches are better decribed by a different code pattern (involving `always @*`).

> Under commit [12132b6](https://github.com/YosysHQ/yosys/commit/12132b6850747aec99715fdfa3184fe3ebefa015), syntax error disappears but proc_dff error still exists. I explore some variants of the code snippet given by @mngr0 .
> 
> 1. variant A
> 
> ```
> always @(posedge rst, negedge clk) begin
>     if (rst) begin
>         q <= 1'b0;
>     end
>     else begin
>         q <= 1'b1;  // or some signal din
>     end
> end
> ```
> 
> got no errors.
> 
> 1. variant B
> 
> ```
> always @(posedge rst, negedge clk) begin
>    q <= rst ? 1'b0 : 1'b1;
> end
> ```
> 
> got **ERROR: Missing edge-sensitive event for this signal!**
> 
> By the way, I join the discussion because I come across variant B and get the error recently. I find no other workarounds to deal with variant B other than changing it to variant A.

This is a different code pattern, different issue.  In this case, well, the Verilog synthesis standard (IEEE 1364.1) actually requires you to write variant A:

"5.2.2.1 Edge-sensitive storage device modeling with asynchronous set-reset
An edge-sensitive storage device with an asynchronous set and/or asynchronous reset is modeled using an always statement whose event list contains edge events representing the clock and asynchronous control variables. Level-sensitive events shall not be allowed in the event list of an edge-sensitive storage device model.
Furthermore, the always statement shall contain an if statement to model the first asynchronous control and optional nested else if statements to model additional asynchronous controls. A final else statement, which specifies the synchronous logic portion of the always block, shall be controlled by the edge control variable not listed in the if and else if statements. The always statement shall be of the form:

```
always @ (posedge <condA> or negedge <condB> or negedge <condC> or ...
posedge <Clock>)
// Any sequence of edge events can be in event list.
if (<condA>) // Positive polarity since posedge <condA>.
// ... <asynchronous logic>
else if (~ <condB>) // Negative polarity since negedge <condB>.
// ... <asynchronous logic>
else if (~ <condC>)
// ... <asynchronous logic>
else // Implicit posedge <Clock>.
// ... <synchronous logic>
```
"

forics, 2020-08-25:
Thanks for your quick and convincing reply. @mwkmwkmwk

povik, 2023-10-02:
It appears this has been addressed, I will go ahead and close the issue.

closed 2023-10-02 by povik