YosysHQ/yosys #668: For loops with a variable in the end-condition

rubund, 2018-10-16:
This works:
`for(i=0;i<10;i++) ...`

but this does not work:
`wire [4:0] variable;`
`for(i=0;i<variable;i++) ..`

It fails with the error:

> ERROR: 2nd expression of generate for-loop is not constant!

This is valid RTL in most other tools, and can be used to save some code implementing more tricky cases.

eddiehung, 2019-02-27:
> This is valid RTL in most other tools, and can be used to save some code implementing more tricky cases.
Can you give me an example of a synthesis tool that supports this?

Background reading:
https://stackoverflow.com/questions/21027464/non-constant-loop-condition-not-supported-for-while
https://stackoverflow.com/questions/35099825/using-a-non-constant-value-inside-while-gives-me-this-error-what-can-i-do
https://groups.google.com/forum/#!topic/comp.lang.verilog/pRr32O7dh10

rubund, 2019-03-06:
> Can you give me an example of a synthesis tool that supports this?

Synopsys Design Compiler at least. Maybe I should have written "most tools I have used" - instead of "most other tools".