YosysHQ/yosys #675: equiv_make breaking for multibit in 0.8

rafaeltp, 2018-10-20:
## Steps to reproduce the issue

input.v
```
module graphtest(A, Y, Z);
  input [3:0] A;
  output [2:0] Y;
  output [1:0] Z;
  assign Y = { A[2:0] & 3'h2 };
  assign Z = A[2:1];
endmodule
```

script:
```
yosys -p "read_verilog graphtest.v; rename -top gold; 
design -stash gold; 
read_verilog graphtest.v; 
rename -top gate; design -stash gate; 
design -copy-from gold -as gold gold; 
design -copy-from gate -as gate gate; 
equiv_make gold gate equiv"
```

## Expected behavior

Equiv circuit generated.

## Actual behavior

Assertion ./kernel/rtlil.h:1307 fails.

log bellow:

```

 /----------------------------------------------------------------------------\
 |                                                                            |
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |                                                                            |
 |  Copyright (C) 2012 - 2018  Clifford Wolf <clifford@clifford.at>           |
 |                                                                            |
 |  Permission to use, copy, modify, and/or distribute this software for any  |
 |  purpose with or without fee is hereby granted, provided that the above    |
 |  copyright notice and this permission notice appear in all copies.         |
 |                                                                            |
 |  THE SOFTWARE IS PROVIDED "AS IS" AND THE AUTHOR DISCLAIMS ALL WARRANTIES  |
 |  WITH REGARD TO THIS SOFTWARE INCLUDING ALL IMPLIED WARRANTIES OF          |
 |  MERCHANTABILITY AND FITNESS. IN NO EVENT SHALL THE AUTHOR BE LIABLE FOR   |
 |  ANY SPECIAL, DIRECT, INDIRECT, OR CONSEQUENTIAL DAMAGES OR ANY DAMAGES    |
 |  WHATSOEVER RESULTING FROM LOSS OF USE, DATA OR PROFITS, WHETHER IN AN     |
 |  ACTION OF CONTRACT, NEGLIGENCE OR OTHER TORTIOUS ACTION, ARISING OUT OF   |
 |  OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.            |
 |                                                                            |
 \----------------------------------------------------------------------------/

 Yosys 0.8+17 (git sha1 11c8a9eb, clang 7.0.0 -fPIC -Os)

-- Running command `read_verilog graphtest.v; rename -top gold; design -stash gold; read_verilog ./graphtest.v; rename -top gate; design -stash gate; design -copy-from gold -as gold gold; design -copy-from gate -as gate gate; equiv_make gold gate equiv' --

1. Executing Verilog-2005 frontend.
Parsing Verilog input from `graphtest.v' to AST representation.
Generating RTLIL representation for module `\graphtest'.
Successfully finished Verilog frontend.
Renaming module graphtest to gold.

2. Executing Verilog-2005 frontend.
Parsing Verilog input from `./graphtest.v' to AST representation.
Generating RTLIL representation for module `\graphtest'.
Successfully finished Verilog frontend.
Renaming module graphtest to gate.

3. Executing EQUIV_MAKE pass (creating equiv checking module).
Presumably equivalent wires: Z_gold (\A_gold [2:1]), Z_gate (\A_gate [2:1]) -> Z
Presumably equivalent wires: Y_gold ($and$graphtest.v:7$1_Y_gold), Y_gate ($and$./graphtest.v:7$2_Y_gate) -> Y
Presumably equivalent wires: A_gold (\A_gold), A_gate (\A_gate) -> A
ERROR: Assert `sig.size() == 1 && sig.chunks().size() == 1' failed in ./kernel/rtlil.h:1307.
```

This started after updating to version 0.8.
Version Yosys 0.7+605 (git sha1 d412b172, clang 6.0.0 -fPIC -Os) does not trigger the assertion (in case that information is useful).

rafaeltp, 2018-10-21:
I was digging around a bit and I think I fixed it in https://github.com/YosysHQ/yosys/pull/676

whitequark, 2018-12-20:
I believe this issue can now be closed.

rafaeltp, 2018-12-20:
Absolutely, thanks for the remainder I haven't been too active the past few weeks.

closed 2018-12-20 by rafaeltp