YosysHQ/yosys #681: fatal error: 'map' file not found

vlsimonk, 2018-10-29:
Hi All,
 I downloaded the yosys files from github by using the below command:
   "git clone https://github.com/cliffordwolf/yosys.git"
  -> then yosys folder downloded
  Later I ran "make config-clang" : Upto now working fine
  Later I ran "make" : Here I am getting below error :
                 "**./kernel/yosys.h:42:10: fatal error: 'map' file not found**"

  Please Help me to solve the above Issue.

Thanks,
Kanth

cliffordwolf, 2018-10-30:
`kernel/yosys.h:42` is `#include <map>`. That is a standard C++ header. Looks like you don't have a proper install of clang.

vlsimonk, 2018-10-30:
Dear Clifford,
   I installed clang properly Still getting the same fatal error. This issue coming only in the "CentOs"  and if i use ubuntu there is no problem.  Please help me to resolve the issue.

Thanks,
Kanth

tomverbeure, 2018-10-30:
Have you tried this:
'sudo apt-get install libc++-dev'

marcusmueller, 2018-10-30:
Here's your bug:

https://github.com/YosysHQ/yosys/blob/master/Makefile#L132

    CXX=clang

needs to be

    CXX=clang+++

`clang` is the C compiler. You need a C++ compiler.

closed 2019-02-27 by eddiehung

dciliske, 2020-06-29:
Can confirm: `make config-clang; make` still triggers this issue.

whitequark, 2020-06-29:
> Here's your bug:

This is technically intended behavior, though it is horrendously wrong. See #2011.

ahmadsb101, 2021-06-13:
Using `CXXFLAGS += -stdlib=libc++ -Os` in the `Makefile` at line 188 or 189 solved the issue for me.