YosysHQ/yosys #700: Feature request: Finer grained control on Clock Enable usage

smunaut, 2018-11-14:
Currently there is the -nodffe option but it's a bit brute force and you don't necessarily want to disable clock-enable usage globally.

Ideally what I'd like is to be able to tell yosys that if a signal would be used as a clock-enable for only 1 (or 2 or N) flip-flops, then don't do it and use mux. This is because in the ice40, clock enable is shared, having a FF with a CE that's only used once has a fairly high cost and makes the placer job much more complicated (leading to placer failing for any design that's somewhere close to to using a large part of the fabric).

In the  dff2dffe pass there seem to be no way to predict how many time a signal was going to be used. So my idea was to run dff2dffe, then do an dff2dffe -unmap  for any dffe where the ce is not used more than 1 time.

Unfortunately, I'm having a hard time counting how many times a CE is used, after the dff2dffe pass, all CE are different 'wires' AFAICT. I tried running an "opt_merge pass before doing the unmap and I see that this pass identifies redudant logic, but the wire names on the 'E' port of the DFFE hasn't changed after that pass so I'm not sure how to identify when 2 wires are in fact the same wire.

smunaut, 2018-11-14:
Ok, managed to make it work, using SigMap to identify connected signals after the opt_merge pass.

Now the question: Do you think it's a worthy feature / good idea ?  If so I can cleanup my code and make a pull request, but if you think it's not a good idea, I don't want to waste time cleaning up something that doesn't stand a chance of being upstreamed.

whitequark, 2018-12-05:
This can be closed now, as #702 is merged.