YosysHQ/yosys #703: ice40 simulation FF primitives don't support init values

olofk, 2018-11-19:
## Steps to reproduce the issue

Run test.sh from [this archive](https://github.com/YosysHQ/yosys/files/2594252/init.tar.gz). This will run an RTL simulation and a simulation on the synthesized netlist

## Expected behavior

Both simulations should output 0 0 1 0 0
## Actual behavior

Synthesized netlist outputs 0 0 z 0 0

whitequark, 2018-12-05:
I can confirm sim/synth mismatch, but I cannot confirm that the issue is init values. iCE40 FFs do not support initialization, so what Yosys does is it inserts inverters before and after (which are then mostly optimized out). You can see this in the synthesized netlist; there is an inverter but no init attributes, as expected:

<details>

```verilog
/* Generated by Yosys 0.8+57 (git sha1 bdab7371, clang 6.0.1-9.2 -fPIC -Os) */

(* top =  1  *)
(* src = "init.v:1" *)
module initvalues(clk, q);
  wire _0_;
  (* src = "init.v:2" *)
  input clk;
  (* src = "init.v:3" *)
  output q;
  (* src = "init.v:5" *)
  wire [3:0] shift;
  (* module_not_derived = 32'd1 *)
  (* src = "/home/whitequark/Projects/yosys/share/ice40/cells_map.v:40" *)
  SB_LUT4 #(
    .LUT_INIT(2'h1)
  ) _1_ (
    .I0(_0_),
    .I1(1'h0),
    .I2(1'h0),
    .I3(1'h0),
    .O(shift[3])
  );
  (* module_not_derived = 32'd1 *)
  (* src = "init.v:7|/home/whitequark/Projects/yosys/share/ice40/cells_map.v:2" *)
  SB_DFF _2_ (
    .C(clk),
    .D(shift[1]),
    .Q(q)
  );
  (* module_not_derived = 32'd1 *)
  (* src = "init.v:7|/home/whitequark/Projects/yosys/share/ice40/cells_map.v:2" *)
  SB_DFF _3_ (
    .C(clk),
    .D(shift[2]),
    .Q(shift[1])
  );
  (* module_not_derived = 32'd1 *)
  (* src = "init.v:7|/home/whitequark/Projects/yosys/share/ice40/cells_map.v:2" *)
  SB_DFF _4_ (
    .C(clk),
    .D(shift[3]),
    .Q(shift[2])
  );
  (* module_not_derived = 32'd1 *)
  (* src = "init.v:7|/home/whitequark/Projects/yosys/share/ice40/cells_map.v:2" *)
  SB_DFF _5_ (
    .C(clk),
    .D(1'h1),
    .Q(_0_)
  );
  assign shift[0] = q;
endmodule
```
</details>

However, looking closer at the traces reveals rather bizarre behavior. The first FF and LUTs have the correct values (i.e. initialization lowering worked), but the output of the LUT is `z` on the first cycle, which is propagated further:

[image]

I don't understand why this happens and it doesn't seem that tweaking `ice40/cells_sim.v` changes much (removing initialization there for FFs results in `x`s, as expected, and nothing else results in any change).

@cliffordwolf Any ideas?

whitequark, 2018-12-05:
Actually, it looks like an issue with the testbench. If I do the following change:
```diff
-   reg clk = 1'b1;
+   reg clk = 1'b0;
   always #1 clk <= ~clk;
```
then I get the expected output:
```
rtl
0
0
0
1
0
synth
0
0
0
1
0
```

olofk, 2018-12-06:
Alright! I think I figured out what's going on. `SB_LUT4` introduces three delta cycles, which means that the FF following the LUT will load an intermediate result from the `SB_LUT4` expressions. Rewriting SB_LUT4 to this seems to fix the problem and I'd argue this is a better solution regardless. But I would need confirmation this doesn't break anything

    module SB_LUT4 (output O, input I0, I1, I2, I3);
    	parameter [15:0] LUT_INIT = 0;
    	assign O = LUT_INIT[{I3,I2,I1,I0}];
    endmodule

daveshah1, 2018-12-06:
Been here when writing the ECP5 sim models! That implementation of a LUT has X-propogation issue, the output will incorrectly be X when any input is X, even if the LUT function and state of other inputs means that the input at X is a don't care. The LUT implementation used currently avoids this with the cascade of muxes (?: won't propogate an X select input if the two values selected between are the same).

olofk, 2018-12-06:
How about

    assign O = LUT_INIT[{(I3 === 1'b1),(I2 === 1'b1),(I1 === 1'b1),(I0 === 1'b1)}];

Not sure if the internal assignments would add a delta cycle though

daveshah1, 2018-12-06:
I think that's gone in the other direction X-propogation wise and will force X inputs to 0 even where they would affect the output based on LUT function (so in those cases the output should be X).

olofk, 2018-12-06:
You wanted cascaded muxes? Well, here's your cascaded muxes from hell :)

	assign O = I0 ?
		   (I1 ?
		    (I2 ?
		     (I3 ?
		      LUT_INIT[15] :
		      LUT_INIT[7]) :
		     (I3 ?
		      LUT_INIT[11] :
		      LUT_INIT[3])) :
		    (I2 ?
		     (I3 ?
		      LUT_INIT[13] :
		      LUT_INIT[5]) :
		     (I3 ?
		      LUT_INIT[9] :
		      LUT_INIT[1]))) :
		   (I1 ?
		    (I2 ?
		     (I3 ?
		      LUT_INIT[14] :
		      LUT_INIT[6]) :
		     (I3 ?
		      LUT_INIT[10] :
		      LUT_INIT[2])) :
		    (I2 ?
		     (I3 ?
		      LUT_INIT[12] :
		      LUT_INIT[4]) :
		     (I3 ?
		      LUT_INIT[8] :
		      LUT_INIT[0])));

It's supposed to be a single-statement rewrite of the original code to avoid the delta cycles. Formatting subject to change. Would this be cool? It's ugly, but the delta cycles are a real problem

olofk, 2018-12-07:
After some further investigations, it seems like this doesn't fix the problem either :/ I'm out of ideas

cliffordwolf, 2018-12-18:
Remember this?  :rofl: 
https://twitter.com/oe1cxw/status/892345863254794241

closed 2018-12-18 by cliffordwolf