YosysHQ/yosys #705: ice40: bad port maps generated for post-synthesis verilog of SB_RAM40_4K

dmsc, 2018-11-20:
Hi,

This verilog code correctly synthesizes an SB_RAM40_4K:

```
module top(
    output reg [7:0] dbr,   // Data bus READ
    input  [7:0] addr,  // Address bus - eight bits
    input  clk          // Clock
    );
    
    reg [7:0] rom_data[0:255];

    always @(posedge clk)
        dbr <= rom_data[addr];

endmodule
```

By using `yosys -q -p 'synth_ice40 -top top -blif top.blif' top.v ` to generate a BLIF file and then `yosys -o top_syn.v top.blif` to generate a post-synthesis  verilog file, the resulting verilog file is invalid, as it contains port declarations like:
```
    .\MASK[0] (_10_),
    .\MASK[10] (_10_),
    .\MASK[11] (_10_),
    .\MASK[12] (_10_),
```

The correct syntax should be ` .MASK( { _10_, _10_, ` etc.

In a more complex design, if I manually fix the MASK, RADDR, RDATA, WADDR and WDATA port declarations I can successfully run a simulation of the synthesized code.

daveshah1, 2018-11-24:
Adding `-wideports` to `read_blif` when reading in the BLIF will merge the ports and thus produce the Verilog output that you expect.

dmsc, 2018-11-24:
> Adding `-wideports` to `read_blif` when reading in the BLIF will merge the ports and thus produce the Verilog output that you expect.

Thanks!

I used `yosys -o top_syn.v -p 'read_blif -wideports  top.blif' ` and works.

Should I close this?

closed 2018-11-24 by daveshah1