YosysHQ/yosys #706: yosys-smtbmc: Problems with "$past()" function in "cover' statements

hotwolf, 2018-11-20:
I'm trying to generate witnesses with yosys-smtbmc and I'm not having much luck with $past() functions in cover statements.

For example, I would like to make sure that a signal ("wb_itr_mon_fsm_busy") can be asserted for two consecutive clock cycles ([see line 408 in the example below](https://github.com/hotwolf/WbXbc/blob/master/bench/verilog/ftb_WbXbc_address_decoder.sv)): 

```
   402     //Cover state transitions in monitor
   403     //==================================
   404     always @(posedge clk_i)
   405         begin
   406  //        cover (wb_itr_mon_fsm_busy & $past(wb_itr_mon_fsm_idle));
   407  //        cover (wb_itr_mon_fsm_idle & $past(wb_itr_mon_fsm_idle));
   408            cover (wb_itr_mon_fsm_busy & $past(wb_itr_mon_fsm_busy));
   409  //        cover (wb_itr_mon_fsm_idle & $past(wb_itr_mon_fsm_busy));
   410
   411            //Enforce witness
   412            cover (wb_itr_mon_fsm_busy & ~rst & req & ack);
   413         end // always @ (posedge clk_i)
```

When I run yosys-smtbmc (through SymbiYosys), it reports that the cover statement in line 408 is unreachable.
However, the cover statement in line 412 is reachable and generates a witness, which satisfies the cover statement in line 408:

```
SBY 21:50:55 [./tools/SymbiYosis/run/cover.WbXbc_address_decoder.default] engine_0: starting process "cd ./tools/SymbiYosis/run/cover.WbXbc_address_decoder.default; yosys-smtbmc --presat --unroll -c --noprogress -t 10 --append 8 --dump-vcd engine_0/trace%.vcd --dump-vlogtb engine_0/trace%_tb.v --dump-smtc engine_0/trace%.smtc model/design_smt2.smt2"
SBY 21:50:55 [./tools/SymbiYosis/run/cover.WbXbc_address_decoder.default] engine_0: ##   0:00:00  Solver: yices
SBY 21:50:56 [./tools/SymbiYosis/run/cover.WbXbc_address_decoder.default] engine_0: ##   0:00:00  Checking cover reachability in step 0..
SBY 21:50:56 [./tools/SymbiYosis/run/cover.WbXbc_address_decoder.default] engine_0: ##   0:00:00  Checking cover reachability in step 1..
SBY 21:50:56 [./tools/SymbiYosis/run/cover.WbXbc_address_decoder.default] engine_0: ##   0:00:00  Checking cover reachability in step 2..
SBY 21:50:56 [./tools/SymbiYosis/run/cover.WbXbc_address_decoder.default] engine_0: ##   0:00:00  Checking cover reachability in step 3..
SBY 21:50:57 [./tools/SymbiYosis/run/cover.WbXbc_address_decoder.default] engine_0: ##   0:00:01  Checking cover reachability in step 4..
SBY 21:50:57 [./tools/SymbiYosis/run/cover.WbXbc_address_decoder.default] engine_0: ##   0:00:01  Checking cover reachability in step 5..
SBY 21:50:57 [./tools/SymbiYosis/run/cover.WbXbc_address_decoder.default] engine_0: ##   0:00:01  Checking cover reachability in step 6..
SBY 21:50:57 [./tools/SymbiYosis/run/cover.WbXbc_address_decoder.default] engine_0: ##   0:00:01  Checking cover reachability in step 7..
SBY 21:50:57 [./tools/SymbiYosis/run/cover.WbXbc_address_decoder.default] engine_0: ##   0:00:01  Checking cover reachability in step 8..
SBY 21:50:58 [./tools/SymbiYosis/run/cover.WbXbc_address_decoder.default] engine_0: ##   0:00:02  Checking cover reachability in step 9..
SBY 21:50:58 [./tools/SymbiYosis/run/cover.WbXbc_address_decoder.default] engine_0: ##   0:00:02  Appending additional step 10.
SBY 21:50:58 [./tools/SymbiYosis/run/cover.WbXbc_address_decoder.default] engine_0: ##   0:00:02  Appending additional step 11.
SBY 21:50:58 [./tools/SymbiYosis/run/cover.WbXbc_address_decoder.default] engine_0: ##   0:00:02  Appending additional step 12.
SBY 21:50:58 [./tools/SymbiYosis/run/cover.WbXbc_address_decoder.default] engine_0: ##   0:00:02  Appending additional step 13.
SBY 21:50:58 [./tools/SymbiYosis/run/cover.WbXbc_address_decoder.default] engine_0: ##   0:00:02  Appending additional step 14.
SBY 21:50:58 [./tools/SymbiYosis/run/cover.WbXbc_address_decoder.default] engine_0: ##   0:00:02  Appending additional step 15.
SBY 21:50:59 [./tools/SymbiYosis/run/cover.WbXbc_address_decoder.default] engine_0: ##   0:00:03  Appending additional step 16.
SBY 21:50:59 [./tools/SymbiYosis/run/cover.WbXbc_address_decoder.default] engine_0: ##   0:00:03  Appending additional step 17.
SBY 21:50:59 [./tools/SymbiYosis/run/cover.WbXbc_address_decoder.default] engine_0: ##   0:00:03  Re-solving with appended steps..
SBY 21:50:59 [./tools/SymbiYosis/run/cover.WbXbc_address_decoder.default] engine_0: ##   0:00:03  Reached cover statement at ftb_WbXbc_address_decoder.sv:412 in step 9.
SBY 21:50:59 [./tools/SymbiYosis/run/cover.WbXbc_address_decoder.default] engine_0: ##   0:00:03  Writing trace to VCD file: engine_0/trace0.vcd
SBY 21:51:00 [./tools/SymbiYosis/run/cover.WbXbc_address_decoder.default] engine_0: ##   0:00:04  Writing trace to Verilog testbench: engine_0/trace0_tb.v
SBY 21:51:00 [./tools/SymbiYosis/run/cover.WbXbc_address_decoder.default] engine_0: ##   0:00:04  Writing trace to constraints file: engine_0/trace0.smtc
SBY 21:51:00 [./tools/SymbiYosis/run/cover.WbXbc_address_decoder.default] engine_0: ##   0:00:04  Checking cover reachability in step 9..
SBY 21:51:00 [./tools/SymbiYosis/run/cover.WbXbc_address_decoder.default] engine_0: ##   0:00:04  Unreached cover statement at ftb_WbXbc_address_decoder.sv:408.
SBY 21:51:00 [./tools/SymbiYosis/run/cover.WbXbc_address_decoder.default] engine_0: ##   0:00:04  Status: FAILED (!)
SBY 21:51:00 [./tools/SymbiYosis/run/cover.WbXbc_address_decoder.default] engine_0: finished (returncode=1)
SBY 21:51:00 [./tools/SymbiYosis/run/cover.WbXbc_address_decoder.default] engine_0: Status returned by engine: FAIL
SBY 21:51:00 [./tools/SymbiYosis/run/cover.WbXbc_address_decoder.default] summary: Elapsed clock time [H:MM:SS (secs)]: 0:00:06 (6)
SBY 21:51:00 [./tools/SymbiYosis/run/cover.WbXbc_address_decoder.default] summary: Elapsed process time [H:MM:SS (secs)]: 0:00:07 (7)
SBY 21:51:00 [./tools/SymbiYosis/run/cover.WbXbc_address_decoder.default] summary: engine_0 (smtbmc) returned FAIL
SBY 21:51:00 [./tools/SymbiYosis/run/cover.WbXbc_address_decoder.default] DONE (FAIL, rc=2)
```
I see the same problem when I try to to implement the $past() function manually with registern.

If you'd like to replay my example, please do the following:

```
git clone https://github.com/hotwolf/WbXbc.git
cd WbXbc
make cover.WbXbc_address_decoder
make debug
```
(Note: "make debug" requires the installation of Perl and the Verilog-Perl library.  Alternatively, run "make debug.list" to see the location of the VCD file and run gtkwave manually)

cliffordwolf, 2019-03-02:
How do you know this is actually reachable within the configured bounds of 10 cycles?

Adding some code that allows `wb_itr_mon_fsm_busy` to be high at any time (and disabling the assertion that would fail as consequence of that) makes the cover statement pass:

```
$ git diff
diff --git a/bench/verilog/ftb_WbXbc_address_decoder.sv b/bench/verilog/ftb_WbXbc_address_decoder.sv
index 1d27c24..0e48f53 100644
--- a/bench/verilog/ftb_WbXbc_address_decoder.sv
+++ b/bench/verilog/ftb_WbXbc_address_decoder.sv
@@ -385,8 +385,8 @@ module ftb_WbXbc_address_decoder
               ~|{wb_itr_mon_fsm_idle, wb_tgt_mon_fsm_idle, wb_pass_through_fsm_idle});
 
         //Busy states must be aligned
-       assert(&{wb_itr_mon_fsm_busy, wb_tgt_mon_fsm_busy, wb_pass_through_fsm_busy} |
-              ~|{wb_itr_mon_fsm_busy, wb_tgt_mon_fsm_busy, wb_pass_through_fsm_busy});
+       // assert(&{wb_itr_mon_fsm_busy, wb_tgt_mon_fsm_busy, wb_pass_through_fsm_busy} |
+       //        ~|{wb_itr_mon_fsm_busy, wb_tgt_mon_fsm_busy, wb_pass_through_fsm_busy});
      end // always @ *
 
    //Cover all target selects
diff --git a/bench/verilog/wb_itr_mon.sv b/bench/verilog/wb_itr_mon.sv
index b2d32fd..70b5e15 100644
--- a/bench/verilog/wb_itr_mon.sv
+++ b/bench/verilog/wb_itr_mon.sv
@@ -208,6 +208,6 @@ module wb_itr_mon
    //==================
    assign tb_fsm_reset = (state_reg === STATE_RESET) ? 1'b1 : 1'b0;
    assign tb_fsm_idle  = (state_reg === STATE_IDLE)  ? 1'b1 : 1'b0;
-   assign tb_fsm_busy  = (state_reg === STATE_BUSY)  ? 1'b1 : 1'b0;
+   assign tb_fsm_busy  = ((state_reg === STATE_BUSY)  ? 1'b1 : 1'b0) || $anyseq(1);
 
 endmodule // wb_itr_mon
```

So I don't think there's anything wrong with `$past`. If you disagree please create an issue with a MCVE that clearly demonstrates that there is a problem. With a large design this is not the case because it isn't obvious at all if the cover statement is reachable within the configured bounds.

closed 2019-03-02 by cliffordwolf

hotwolf, 2019-03-04:
Thanks for looking into this issue.  I guess the test case I've prepared appears to be more complex, than it really is. All the [design ](https://github.com/hotwolf/WbXbc/blob/master/rtl/verilog/WbXbc_address_decoder.v) does, is pass through a set of wishbone signals. It doesn't contain any sequential logic. The test bench contains two cover statements:  

- The first one "cover (wb_itr_mon_fsm_busy & $past(wb_itr_mon_fsm_busy));" is reported as unsatisfiable.

- The second one "cover (wb_itr_mon_fsm_busy & ~rst & req & ack);" produces a witness which contradicts the first result. (wb_itr_mon_fsm_busy is asserted for two consecutive clock cycles). Here is a [screenshot](https://hotwolf.github.io/WbXbc/images/issue706.png).

I don't know how to simplify the test case.  The make file I've provided let's you reproduce the problem with a single command  ("make cover.WbXbc_address_decoder").
 I'll re-post the issue one more time, in case you do want to follow up on why Yosys is reporting these contradictory results.

PS: The assertions you've removed are required for running proofs with k-induction.

ZipCPU, 2019-03-05:
See the answer given for #853 .  The tool is working as intended, and it is the "append" statement that is confusing you.  One more clock is necessary to successfully find the missing trace.

hotwolf, 2019-03-05:
You are right, I did not consider that part of the trace consists appended cycles. Thanks for pointing this out to me.